iNavFlight/inav-configurator #996: Omnibus F4 Pro V2 UART3

ltwin8, 2020-05-16:
Hello community,

I have an omnibus pro v2 with baro in a Wing.

GPS to UART1, vtx on Softseial (both confirmed working)

And also receiver (RX TX lines) to UART3, the receiver is not working. 
I do not intend to use the I2C Bus that is shared with those pins.

Is the barometer maybe conflicting with UART3?

DzikuVx, 2020-05-17:
This board has no inverters on UART3. Depending on the veriant it's either UART1 or 6.

closed 2020-05-17 by DzikuVx

ltwin8, 2020-05-17:
Update: the receiver works, but VERY slow and inconsistent (like an update Avery 10-60 seconds), all other peripherals work and receiver is also tested on another board. 

I use this board https://www.banggood.com/FLIP32-F4-OMNIBUS-V2-PRO-Controller-Board-Kit-Built-In-OSD-Module-Current-Sensor-p-1221793.html?rmmds=search&cur_warehouse=CN