iNavFlight/inav-configurator #997: 2.5.0 RC2 hanging on Mac

flythebluesky, 2020-05-16:
Configurator 2.5.0 RC2 on Mac 10.15.4 hangs/non-responsive during initial setup w/KAKUTEF7. 

1. I tried the new standalone configurator version, and also a local build. See logs below. 
2. The status bar at the bottom shows the "Drop Ratio" bouncing around from 0% up to ~25%.
3. Things seem to go off when reaching the configurator tab but who knows if that is a coincidence. 
4. I am unable to provide "diff all" as this command hangs. 

For config I start at the top and move down except for calibration which requires the battery to be attached. I enable motor output and set DSHOT600.

2020-05-16 @ 10:24:14 -- Running - OS: MacOS, Chrome: 78.0.3904.87, Configurator: 2.5.0
2020-05-16 @ 10:24:15 -- Loaded release information from GitHub.
2020-05-16 @ 10:24:17 -- Serial port successfully opened with ID: 1
2020-05-16 @ 10:24:17 -- MultiWii API version received - 1.43.0
2020-05-16 @ 10:24:17 -- This firmware variant is not supported. Please upgrade to INAV firmware. Use CLI for backup before flashing. CLI backup/restore procedure is in the documention.
2020-05-16 @ 10:24:23 -- Serial port successfully closed
2020-05-16 @ 10:25:04 -- USB device successfully opened with ID: 1
2020-05-16 @ 10:25:04 -- Detected device with total flash size 1024 kiB
2020-05-16 @ 10:25:04 -- Read protection not active
2020-05-16 @ 10:25:12 -- Erased 1024 kB of flash successfully
2020-05-16 @ 10:25:39 -- USB device successfully closed
2020-05-16 @ 10:26:15 -- Serial port successfully opened with ID: 3
2020-05-16 @ 10:26:15 -- MultiWii API version received - 2.4.0
2020-05-16 @ 10:26:15 -- Flight controller info, identifier: INAV, version: 2.5.0
**2020-05-16 @ 10:26:15 -- Running firmware released on: May 9 2020 14:36:53**
2020-05-16 @ 10:26:15 -- Board: KTF7, version: 0
2020-05-16 @ 10:26:15 -- Unique device ID received - 0x1f00363236510530383430
2020-05-16 @ 10:26:15 -- PWM output is disabled. Motors and servos will not work. Use Outputs tab to enable!
2020-05-16 @ 10:26:15 -- Mixer not configured. Use Mixer tab to set it up!
2020-05-16 @ 10:26:23 -- EEPROM saved
2020-05-16 @ 10:26:23 -- Device - Rebooting
2020-05-16 @ 10:26:23 -- Serial port successfully closed
2020-05-16 @ 10:26:30 -- Serial port successfully opened with ID: 4
2020-05-16 @ 10:26:30 -- MultiWii API version received - 2.4.0
2020-05-16 @ 10:26:30 -- Flight controller info, identifier: INAV, version: 2.5.0
2020-05-16 @ 10:26:30 -- Running firmware released on: May 9 2020 14:36:53
2020-05-16 @ 10:26:30 -- Board: KTF7, version: 0
2020-05-16 @ 10:26:30 -- Unique device ID received - 0x1f00363236510530383430
2020-05-16 @ 10:26:30 -- PWM output is disabled. Motors and servos will not work. Use Outputs tab to enable!
2020-05-16 @ 10:26:30 -- Mixer not configured. Use Mixer tab to set it up!
2020-05-16 @ 10:26:58 -- EEPROM saved
2020-05-16 @ 10:26:58 -- Device - Rebooting
2020-05-16 @ 10:26:58 -- Serial port successfully closed
2020-05-16 @ 10:27:05 -- Serial port successfully opened with ID: 5
2020-05-16 @ 10:27:05 -- MultiWii API version received - 2.4.0
2020-05-16 @ 10:27:05 -- Flight controller info, identifier: INAV, version: 2.5.0
2020-05-16 @ 10:27:05 -- Running firmware released on: May 9 2020 14:36:53
2020-05-16 @ 10:27:05 -- Board: KTF7, version: 0
2020-05-16 @ 10:27:05 -- Unique device ID received - 0x1f00363236510530383430
2020-05-16 @ 10:27:06 -- Mixer not configured. Use Mixer tab to set it up!
2020-05-16 @ 10:27:33 -- EEPROM saved
2020-05-16 @ 10:27:33 -- Device - Rebooting
2020-05-16 @ 10:27:33 -- Serial port successfully closed
2020-05-16 @ 10:27:40 -- Serial port successfully opened with ID: 6
2020-05-16 @ 10:27:40 -- MultiWii API version received - 2.4.0
2020-05-16 @ 10:27:40 -- Flight controller info, identifier: INAV, version: 2.5.0
**2020-05-16 @ 10:27:40 -- Running firmware released on: May 9 2020 14:36:53**
2020-05-16 @ 10:27:40 -- Board: KTF7, version: 0
2020-05-16 @ 10:27:40 -- Unique device ID received - 0x1f00363236510530383430
2020-05-16 @ 10:27:40 -- Mixer not configured. Use Mixer tab to set it up!
2020-05-16 @ 10:29:05 -- Serial port successfully closed
2020-05-16 @ 10:29:05 -- Failed to open serial port
2020-05-16 @ 10:29:15 -- Serial port successfully opened with ID: 7
2020-05-16 @ 10:29:15 -- MultiWii API version received - 2.4.0
2020-05-16 @ 10:29:15 -- Flight controller info, identifier: INAV, version: 2.5.0
2020-05-16 @ 10:29:15 -- Running firmware released on: May 9 2020 14:36:53
2020-05-16 @ 10:29:15 -- Board: KTF7, version: 0
2020-05-16 @ 10:29:15 -- Unique device ID received - 0x1f00363236510530383430
2020-05-16 @ 10:29:15 -- Mixer not configured. Use Mixer tab to set it up!
**2020-05-16 @ 10:30:13 -- You need to upgrade your firmware before you can use the Configuration tab.**
2020-05-16 @ 10:30:17 -- Serial port successfully closed
2020-05-16 @ 10:30:21 -- Failed to open serial port
2020-05-16 @ 10:30:28 -- Serial port successfully opened with ID: 8
2020-05-16 @ 10:30:30 -- MultiWii API version received - 2.4.0
2020-05-16 @ 10:30:30 -- Flight controller info, identifier: INAV, version: 2.5.0
2020-05-16 @ 10:30:30 -- Running firmware released on: May 9 2020 14:36:53
2020-05-16 @ 10:30:30 -- Board: KTF7, version: 0
2020-05-16 @ 10:30:30 -- Unique device ID received - 0x1f00363236510530383430
2020-05-16 @ 10:30:30 -- Mixer not configured. Use Mixer tab to set it up!
2020-05-16 @ 10:30:32 -- CLI mode detected

maponarooo, 2020-05-18:
Where is MAC configurator?

flythebluesky, 2020-05-18:
> Where is MAC configurator?

It's mentioned right at the top of the documentation: https://inavapp.com

DzikuVx, 2020-05-19:
The https://inavapp.com is not really a INAV project and INAV team is not providing any support for it at the moment

closed 2020-05-19 by DzikuVx

flythebluesky, 2020-05-19:
@DzikuVx understood, thanks. Note that it doesn't work when run directly from source either. This means there is no support for Mac at all :-( I will try to look into it sometime.