iNavFlight/inav-configurator #1000: Mission valid is unclear

Etruscian, 2020-05-22:
When planning a mission, I think it is really unclear that it shows: mission valid with a red cross until I upload it. 
Diving into the code, I can't easily tell why it isn't valid until I upload it, but a suggestion from my side would be to add a tooltip stating the reason or changing the icon into text explaining why it is not valid.

Mission valid means to me that the mission itself is not valid, i.e. flying below ground level or flying with a negative velocity. If I'm putting my first waypoint down with the default settings and seeing mission valid with a cross without any explanation, I get confused to why that is so.

After creating a mission, saving it to the FC and then removing all the points, the mission valid shows a green icon, confusing me even more. If I'm not planning a mission at the moment, why is it valid now, but not when I open Inav for the first time?

PredySoft, 2021-11-04:
I agree  - I just did plan the mission and to me it seems it indicates rather if the mission is uploaded to Fligth Controller than any validity of the mission itself. So I would rather expect something like "Mission ready:" or "Mission uploaded to FC:" Or "Mission saved:"

VancouverUmbrella, 2022-05-07:
I'm happy that I found this bug report. 

Being new to iNav, I was very confused by the fact that missions showed as invalid. Like other above, I interpreted "Mission invalid" to mean just that i.e. there was something wrong with the mission. 

In fact, it turns it out it meant something like 'Mission not saved to eeprom' or 'Mission not uploaded to FC' in my use case.

b14ckyy, 2025-01-19:
collected in https://github.com/iNavFlight/inav-configurator/issues/1181

closed 2025-01-19 by b14ckyy