iNavFlight/inav-configurator #1001: Missing feature: BLE support

marianomd, 2020-05-22:
BLE doesn't have RFCOMM support, so it would be very useful to configure iNav through HC-08 or HM-10 modules or directly on FCs like SpeedyBee F4/F7, Racerstar MELO F4, Diatone Mamba F722S, and other FCs that have CC2541 chip.

Chrome has support for BLE: https://googlechrome.github.io/samples/web-bluetooth/
I managed to discover the device with one example from that page.

I guess we could take the implementation details from a project like this: https://github.com/monteslu/ble-serial

It seems it should be straightforward. I will investigate further.

tonyyng, 2020-06-14:
I would find this very useful.  I found it mentioned in this [issue](https://github.com/iNavFlight/inav-configurator/issues/2), but it doesn't appear it went anywhere at that point.

elepete, 2020-06-30:
Another thought on BLE (HM-10 / HM-11 like modules). As rfcomm is not supported there is a way to get a "pts" device linked to serial communcation of the BLE module. Please see this link:
https://github.com/Jakeler/ble-serial
This is an easy way to get a "serial" for BLE in Linux (e.g. as /dev/pts/2). I have tested that under Debian Buster, works fine and stable. Only issue is that you cannot open any device that is not dialout or a full tty. Terminal programs perform just fine on a 'pts' and communicate well via BLE to the MSP serial port on the FC.
Issue is that INAV_Configurator is not accepting or really creating an error opening such 'pts' device instead of a 'tty'. If the configurator could be changed that would be another way of having BLE serial support.

If all that works I would add a BLE to my copter, and have a logic switch to enable it in disarmed state only. In that way the USB cable is only needed in super rare cases + phone is always a good option to do changes if needed.

BLE support in Configurator would indeed be a real plus.

elepete, 2020-07-04:
Under Linux it works with the ble-serial.
A thanks to Jakeler who created the ble-serial and looked behind things:
https://github.com/Jakeler/ble-serial/issues/6#issuecomment-653365093

michaelvanderhoek, 2021-05-25:
ANy updates on this topic?

MrD-RC, 2022-11-26:
Fixed with https://github.com/iNavFlight/inav-configurator/issues/1436

closed 2022-11-26 by MrD-RC