iNavFlight/inav-configurator #1002: Receiver Mode is not apearing in iNAV-CONFIGURATOR-2.5RC2

WohaibHasan, 2020-05-22:
after I have erased the FC for several time with iNAV2.5RC1 , the (Receiver Mode) on the (Configuration) tab is not there any more! i have already deleted the (C:\Users\MyUserName\AppData\Local\inav-configurator) folder then redounded the INAV Configurator 2.5.0 RC2 again. still the same issue. i have tested this in 3 FC's (F7's), but no lock. restarted the PC , deleted the folder, then redownloaded the Config2.5RC2 again, but the same issue !!

the (Receiver Mode) shows up in the CONFIGURATOR 2.4.1 normally without any issues!

DzikuVx, 2020-05-22:
Receiver configuration was moved to the Receiver tab where is should have been from the beginning

closed 2020-05-22 by DzikuVx

WohaibHasan, 2020-05-22:
OMG, :) . yes. i have seen it now. thanks