iNavFlight/inav-configurator #1011: Units for all variables

kibinek, 2020-05-28:
## Current Behavior

In many cases there are not shown units at variables in INAV GUI and CLI. 
## Desired Behavior

INAV GUI and CLI shall have relevant units for all variables, in square brackets [] of course.
## Suggested Solution

Please add relevant unit to each variable in both GUI and CLI.
## Who does this impact? Who is this for?

It shall be done by INAV developers. For INAV regular users.
## Additional context

I believe developers know what unit each variable has, but for regular user sometimes it's unclear if a variable is either in [Hz] or [%] or...

kibinek, 2020-05-28:
Thank you Konstantin. But I believe having units mentioned in CLI is as important as in GUI. But truly, it's more important in GUI, since GUI variables are visible immediatelly, whereas in CLI on demand only.

zoran-grujic, 2022-04-10:
It is not mentioned here but in many GUI entry boxes (not in all of them) now we have units, but then we also have units in the box description inside [ ] but this units are rarely the same! 

For example: Advanced Tuning>Threshold velocity [cm/s]  - but in the input box is shown Km/h. 
This is confusing! 
Also it should be km/h (not Km/h) according to SI system of units. In this specific case m/s is more suitable, but this is not an issue - use m/s or km/h and stick to it across the entire app!

MrD-RC, 2022-04-10:
> It is not mentioned here but in many GUI entry boxes (not in all of them) now we have units, but then we also have units in the box description inside [ ] but this units are rarely the same!
>
> For [...]

Fixed in https://github.com/iNavFlight/inav-configurator/pull/1416

closed 2022-11-26 by MrD-RC