iNavFlight/inav-configurator #1013: Cannot open Configuration tab in iNav Configurator 2.5 RC2

pepjo, 2020-05-29:
I cannot load the configuration tab in iNav configurator.

[image]

In the console I can see the following output:
```
The specified value "NaN" cannot be parsed, or is out of range.
(anonymous) @ script.js:92
each @ script.js:90
each @ script.js:90
val @ script.js:92
(anonymous) @ script.js:1503
each @ script.js:90
each @ script.js:90
a.fn.val @ script.js:1503
process_html @ script.js:33718
r @ script.js:18888
i._settlePromiseFromHandler @ script.js:18887
i._settlePromise @ script.js:18887
i._settlePromise0 @ script.js:18887
i._settlePromises @ script.js:18887
r._drainQueue @ script.js:18886
r._drainQueues @ script.js:18886
drainQueues @ script.js:18886
Promise.then (async)
r @ script.js:18887
r._queueTick @ script.js:18886
s @ script.js:18886
p.hasDevTools.r.settlePromises @ script.js:18886
i._fulfill @ script.js:18887
i._resolveCallback @ script.js:18887
(anonymous) @ script.js:18887
(anonymous) @ script.js:27596
self.processData @ script.js:21462
_dispatch_message @ script.js:27469
read @ script.js:27448
read_serial @ script.js:25786
```
```
script.js:92 The specified value "NaN" cannot be parsed, or is out of range.
(anonymous) @ script.js:92
each @ script.js:90
each @ script.js:90
val @ script.js:92
(anonymous) @ script.js:1503
each @ script.js:90
each @ script.js:90
a.fn.val @ script.js:1503
process_html @ script.js:33719
r @ script.js:18888
i._settlePromiseFromHandler @ script.js:18887
i._settlePromise @ script.js:18887
i._settlePromise0 @ script.js:18887
i._settlePromises @ script.js:18887
r._drainQueue @ script.js:18886
r._drainQueues @ script.js:18886
drainQueues @ script.js:18886
Promise.then (async)
r @ script.js:18887
r._queueTick @ script.js:18886
s @ script.js:18886
p.hasDevTools.r.settlePromises @ script.js:18886
i._fulfill @ script.js:18887
i._resolveCallback @ script.js:18887
(anonymous) @ script.js:18887
(anonymous) @ script.js:27596
self.processData @ script.js:21462
_dispatch_message @ script.js:27469
read @ script.js:27448
read_serial @ script.js:25786
```
```
script.js:18886 Unhandled rejection TypeError: Cannot read property 'tick' of undefined
    at HTMLSelectElement.<anonymous> (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:33776:86)
    at HTMLSelectElement.dispatch (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:91:6466)
    at HTMLSelectElement.r.handle (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:91:3241)
    at Object.trigger (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:91:5574)
    at HTMLSelectElement.<anonymous> (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:91:11542)
    at Function.each (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:90:2882)
    at n.fn.init.each (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:90:847)    at n.fn.init.trigger (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:91:11518)
    at n.fn.init.n.fn.<computed> [as change] (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:92:5101)
    at process_html (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:33821:18)
    at r (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:18888:7722)
    at i._settlePromiseFromHandler (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:18887:13044)
    at i._settlePromise (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:18887:13847)
    at i._settlePromise0 (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:18887:14548)
    at i._settlePromises (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:18887:15878)
    at r._drainQueue (chrome-extension://hpbekhcbpgdlggfaajdbjjmhpogcdhod/build/script.js:18886:3148)
```

DzikuVx, 2020-05-29:
@pepjo which exact version on INAV is flashed to the board and what exact board it is?

pepjo, 2020-05-29:
Hi @DzikuVx it's a new Kakute F7 AIO v1.5. I tried running inav 2.5 RC1 and RC2, they both have the same problem.

pepjo, 2020-05-29:
I flashed the flight controller again and now it works... sorry for wasting your time.

closed 2020-05-29 by pepjo