iNavFlight/inav-configurator #1024: macOS standalone app doesn't see bluetooth wireless com ports while chrome app does

gegtor, 2020-06-18:
New app from inavapp.com doesn't see the Bluetooth serial ports while the standard one does

[image]
inavapp.com app
[image]
Standard configurator
[image]

gegtor, 2020-06-18:
macOS 10.15.5
MBP 15" touchbar 2019
HC-06 module

fiam, 2020-06-27:
@gegtor Thanks for the heads up. The macOS app is filtering all unknown port types, in order to avoid cluttering the display. Right now, it only shows serial ports associated to USB devices or BLE devices. I've just ordered some HC-06 modules (which use BT classic) to be able to test this.

Airwide, 2020-07-24:
I have the same issue.

MacBook Pro (Retina, 13-inch, Late 2013)
macOS 10.15.5
HC-06 module
INAV APP Version 0.9.10 (8)