iNavFlight/inav-configurator #1057: iFlight SucceX-E F4 Stack is listed on Release page but no target

ircmgr, 2020-09-01:
iFlight SucceX-E F4 Stack can be seen in the recommended hardware list but while trying to flash only 
IFLIGHTF4- TWIG and IFLIGHTF7-TWING targets exist ( tried the F4 and it doesnt work )

BrianKoch1, 2020-12-02:
Is there a solution for this board yet?

theuglycamel, 2020-12-28:
Also looking for this target.  The V1 board uses the BETAFLIGHT_F4 target but the v2.1 uses IFLIGHT_SUCCEX-E F4.  The V2.1 is a much more iNav friendly board with baro and i2c.  Please advise.

bakwc, 2021-03-05:
Same here. Is it possible to run inav on "iFlight SucceX-E F4 V2.1" (SucceX-E F405)?

ryanort, 2021-03-09:
> Same here. Is it possible to run inav on "iFlight SucceX-E F4 V2.1" (SucceX-E F405)?

+1

ucflumm, 2021-03-13:
+1 Tried flashing it but it only shows 3 UART and exhibits strange behaviour i'm guessing i got the wrong target.

ucflumm, 2021-03-13:
I'll attempt to make a target, but I've never done it for iNAV/betaflight or anything. I've done it for 3D printer boards using STM32 so I'll give it a shot. 
iflightf4_succexe/target.c
                            target.h

Does anyone know if there are other references to these (in iNAV/configurator) I need to add/check anywhere. I could find it with time but I think it would be easier to ask.

Sorry if I'm asking noob questions; I just feel it's alot better to help write it and test instead of asking.

ivan586lp, 2021-03-16:
I managed to modify the target BETAFLIGHT_F4 (src / main / target / BETAFLIGHTF4 / target.h)

Yesterday I took off on a drone (on version 2.6.0), did not test for a long time, but the GPS holds the position, the flight is stable, the acro is working, all functions are working (barometer, GPS, Compass)

If you try, please be careful :)

Link to archive (compiled firmware hex file and target.h):

https://drive.google.com/file/d/1-wDzZs2W_LAJGBvbQVb52oV7JjGyI9DB/view?usp=sharing

ucflumm, 2021-03-16:
> I managed to modify the target BETAFLIGHT_F4 (src / main / target / BETAFLIGHTF4 / target.h) for iFlight SucceX-E F4 V2.1
> 
> Yesterday I took off on a drone (on version 2.6.0), did not test for a long time, but the GPS holds the position, the flight is stable, the acro is working, all functions are working (barometer, GPS, Compass)
> 
> If you try, please be careful :)
> 
> Link to archive (compiled firmware hex file and target.h):
> 
> https://drive.google.com/file/d/1-wDzZs2W_LAJGBvbQVb52oV7JjGyI9DB/view?usp=sharing

Awesome save me heaps of time I'll try it out this weekend!

bakwc, 2021-03-26:
Successfully flashed, have not tested yet.
I am missing barometer values (may be my board without barometer, or it just not detected, I can't setup it in beta-flight too).
Any ideas how to make baro works?

ucflumm, 2021-03-26:
Is it showing up as red or is it greyed out?

bakwc, 2021-03-27:
> Is it showing up as red or is it greyed out?

Red

ucflumm, 2021-03-28:
Have you tried a full power off. Mine was red when I connected to usb first. Disconnect usb and power. Then connect battery after that plug in the usb to see if that fixes it.

TarasTadva, 2021-03-30:
Hey! Please explain how to add a target to the configurator yourself? I successfully flashed the flight controller with the suggested HEX file, but did not use the target at all. This is mistake?

ucflumm, 2021-03-30:
The target is used to create the hex file. Please be careful flashing verify it works first.

nwpulver, 2021-04-20:
> Awesome save me heaps of time I'll try it out this weekend!

I am having an issue with this target  - ESC not talking to FC. Not sure what the deal is because it works on betaflight. Anyone else have this problem?

ucflumm, 2021-04-20:
> > > I managed to modify the target BETAFLIGHT_F4 (src / main / target / BETAFLIGHTF4 / target.h) for iFlight SucceX-E F4 V2.1
> > > Yesterday I took off on a drone (on version 2.6.0), did not test for a long time, but the GPS holds the position, the flight is stable, the acro is working, all functions are working (barometer, GPS, Compass)
> > > If you try, please be careful :)
> > > Link to archive (compiled firmware hex file and target.h):
> > > https://drive.google.com/file/d/1-wDzZs2W_LAJGBvbQVb52oV7JjGyI9DB/view?usp=sharing
> > 
> > 
> > Awesome save me heaps of time I'll try it out this weekend!
> 
> I am having an issue with this target - ESC not talking to FC. Not sure what the deal is because it works on betaflight. Anyone else have this problem?
> 
> to elaborate: I am not getting the second set of beeps when plugging in battery. I cannot move motors with outputs tab. In betalfight this is no problem.

Did you set the motor protocol dshot600 ? Won't beep if its not set i think.

theuglycamel, 2021-04-25:
Successfully tested 04/24/21 on fixed wing.  Thank you for the conversion!

nwpulver, 2021-04-26:
> Did you set the motor protocol dshot600 ? Won't beep if its not set i think.

I did. I now notice that I get the second set of beeps on reboot as it is rebooting but it does not happen otherwise. Using BLHeliS and bluejay firmware

lavachemist, 2021-04-27:
@digitalentity did you know that the release tag 2.6.1 for INAV configurator recommends an FC that has no INAV target?

https://github.com/iNavFlight/inav-configurator/releases/tag/2.6.1

jandaliba, 2021-05-05:
Same here... I bought it based on recomended HW and now I do not have a target... Just perfect...

mimoklepes, 2021-05-21:
I've complied the target using target.h from ivan586lp with INAV 3.0.0 RC1 and tested today by doing 2 battery packs on a 5" quad (Nazgul5 + BN-880 GPS). So far everything seems to work great.

EchoReeper, 2021-05-31:
> I've compiled the target using target.h from ivan586lp with INAV 3.0.0 RC1 and tested today by doing 2 battery packs on a 5" quad (Nazgul5 + BN-880 GPS). So far everything seems to work great.

I just got the shared file ivan586lp (thank you sir) to work, is there a way you can share the 3.0 version please?

mimoklepes, 2021-05-31:
> I just got the shared file ivan586lp (thank you sir) to work, is there a way you can share the 3.0 version please?

sure, [here you go](https://drive.google.com/file/d/1IKo4rn5m6UVKGCpeJKJxkf_ezd2D1M0S/view?usp=sharing)

It is built from 3.0.0 RC1. In the meantime I've tested it for a couple of hours of flight time and everything was fine, so I'm not going to bother rebuilding it from RC2, I'll wait for the final version. 

Just one warning: The very first attempt almost ended with a disaster - the quad started to add throttle by itself, I had to quickly disarm it mid-air (it was already some 3 meters above the ground), otherwise it would just fly away. I guess the problem was that I did "diff all" on the 2.6.0 version and applied everything blindly to 3.0.0.

The second time I started from scratch using a preset from configurator and only copied the "harmless" parts of the diff like OSD, modes, rates, etc. Zero problems since then.

EchoReeper, 2021-05-31:
> sure, [here you go](https://drive.google.com/file/d/1IKo4rn5m6UVKGCpeJKJxkf_ezd2D1M0S/view?usp=sharing)
>
> It is built from 3.0.0 RC1. In the meantime I've tested it for a couple of hours of flight [...]

Thanks so much! I just loaded and the motors spin (I didn't copy and paste anything over, thanks' for the tip). I will test and report back if anything happens, etc.. off to test

Dave4675, 2021-06-23:
Hi everybody, thanks to 

> I managed to modify the target BETAFLIGHT_F4 (src / main / target / BETAFLIGHTF4 / target.h) for iFlight SucceX-E F4 V2.1
>
> Yesterday I took off on a drone (on version 2.6.0), did not test for a long [...]

Thanks, I used the target file to compile inav 3.0 :) 
I'm posting the zip file that contain the hex file.

[inav_3.0.0_IFLIGHT_SUCCEX_E_F4_2_1.hex.zip](https://github.com/iNavFlight/inav-configurator/files/6701757/inav_3.0.0_IFLIGHT_SUCCEX_E_F4_2_1.hex.zip)

Ajaxwars, 2021-07-12:
Доброго времени суток всем.
Спасибо за работу над этой прошивкок.
Но есть проблема, которую не могу решить.
Прошивка заливается нормально, всё настраивается, иду полетать, всё работает кроме правого стика, ролл и питч не отвечают.
На компьютере  в программе всё работает значения меняются.
А в полёте только взлетает и нельзя никуда повернуть.
Аппа радиомастер tx16s.
Бетафлайт всё работает как надо, Летаю.
Помогите пожалуйста.

olegshty, 2021-08-04:
> Hi everybody, thanks to
> 
> > I managed to modify the target BETAFLIGHT_F4 (src / main / target / BETAFLIGHTF4 / target.h) for iFlight SucceX-E F4 V2.1
> > Yesterday I took off on a drone (on version 2.6.0), did not test for a long time, but the GPS holds the position, the flight is stable, the acro is working, all functions are working (barometer, GPS, Compass)
> > If you try, please be careful :)
> > Link to archive (compiled firmware hex file and target.h):
> > https://drive.google.com/file/d/1-wDzZs2W_LAJGBvbQVb52oV7JjGyI9DB/view?usp=sharing
> 
> Thanks, I used the target file to compile inav 3.0 final :)
> I'm posting the zip file that contain the hex file.
> 
> [inav-iflight-succex-e-v2.1.zip](https://github.com/iNavFlight/inav-configurator/files/6701769/inav-iflight-succex-e-v2.1.zip)

Thank you for a great job, but there is a mistake in resources:
In your HEX B05: LED1 OUT
But for real LED1 OUT is B06
Could you please correct this...?

Mcbeen, 2021-12-20:
> Successfully tested 04/24/21 on fixed wing. Thank you for the conversion!

@theuglycamel That sounds awesome. Would love to be able to do that. But where did you connect your servos? Motor outputs 3 & 4? Did you just use Dave4675's target or..? Would love to know. Thanks already!

MrD-RC, 2022-11-26:
Stale closing. For official targets to be added to INAV. Manufacturers should follow [these rules](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md). If a target for a flight controller does not exist. Contact the manufacturer and as them to get an INAV target made and approved.

closed 2022-11-26 by MrD-RC