iNavFlight/inav-configurator #1058: Rover brushed esc wants to full throttle upon power up

tnap1979, 2020-09-01:
mateksys f722 wing edition flight controller,
Matkesys m8q-5883 gps/compass,
128x64 oled display,
Crossfire diversity nano receiver
Tamiya teu - 106bk dual brushed esc with dual 45t 2magnet/pole motors
Integy scx10 custom 4x4 crawler frame
Molicel 21700 4200mah 45a li-ion in 2s2p

Hardware all works perfectly but as soon as I power on the esc, it beeps twice and seems to go full throttle
Running latest inav release from configuration app on surface pro 6 w/windows 10 pro ..
I cant configure the esc either, seems like inav isn't allowing the radio information to pass thru to the esc to try to calibrate  it

packetpilot, 2024-01-12:
What sort of TX are you running -- is it a stick radio? If so:
it's possible that your full stick down (on throttle) isn't registering mentally as "WOT in reverse" in your brain. Should this make sense, you can set up a mix w/ two throttles
- one w/ 50% weight and 50% offset
- another with -50% weight and -50% offset
  - be sure this second one (and _only_ this second one) is conditional on a switch being thrown (I personally like SA ⬇️ )
- both of these should output to your throttle channel

Do some testing w/ the pinion off, or disengaged from the spur at least.

Apologies if my hunch is an irrelevant tangent. An aside, INAV is _really_ not set up to make rover stuff straightforward (especially when it comes to nav params & the very low signal-to-noise ratio the configurator presents us with, despite it "knowing" that a rover is the intended use case).

sensei-hacker, 2024-01-12:
If you have the motors / ESC on a *servo* output, and you have it set to bidirectional, that would explain it. If that's the case, changes will be needed.

closed 2025-01-19 by b14ckyy