iNavFlight/inav-configurator #1070: INAV does not allow reconfiguration of accelerometer.

cfanyu2010, 2020-09-27:
I configured it several months ago.  I press configure.  It gives me the standard message.  I press ok.  Then all 6 positions are checked and it shows me "Accelerometer calibration finished".  In the log at the top left, Accelerator calibration started and finished in two seconds.  

I appreciate your help.  I am working with a Matek F765.

CaptDen, 2020-10-13:
I have the same issue. I have the Matek 722-SE. I have also tried to reset all the acc entiries to there lowest settings to if this would permit a reset of the calibration process.

cfanyu2010, 2020-10-13:
to reset the calibration and start from scratch you can use these stock
telemetry data and put them into CLI and save them. This will reset the
telemetry and you can start again: set acczero_x = 0 set acczero_y = 0 set
acczero_z = 0 set accgain_x = 4096 set accgain_y = 4096 set accgain_z =
4096 save

JohnDevine, 2020-11-02:
Same here .. Matek 722-SE. Maybe an issue with this particular controller?

zacool, 2021-02-09:
Same here .. Matek 722-SE.
Any solution ?

cfanyu2010, 2021-02-09:
start over

JohnDevine, 2021-02-10:
I found a solution. My gps/compass is mounted upside down in a tower. I I hold the quad level and try to calibrate it will not work. If I first turn it upside down and try to calibrate it works perfectly.

zacool, 2021-02-10:
> start over
> […](#)

start over what?

zacool, 2021-02-10:
> I found a solution. My gps/compass is mounted upside down in a tower. I I hold the quad level and try to calibrate it will not work. If I first turn it upside down and try to calibrate it works perfectly.

Yes it worked. 
Thank you very much.
But why i must turn up side down to begin?
Can you explain it to me?
I thought that accelerometer is on flight controller.

JohnDevine, 2021-02-11:
sorry zacool, I have no idea why. The only connection I can see is that the compass is also in the tower and the calibration of the accel code is somehow tangled up in that. I hope it helps the developers.

zacool, 2021-02-11:
> sorry zacool, I have no idea why. The only connection I can see is that the compass is also in the tower and the calibration of the accel code is somehow tangled up in that. I hope it helps the developers.

ok thanks anyway.

petertorelli, 2021-09-06:
I'm seeing this too. The FC keeps returning 0x3F during the MSP_CALIBRATION_DATA (loadCalibrationData) MSP chain function saying all six positions are calibrated. Instead of ignoring this value, the GUI sees that all six bits are set and does not invoke the calibration routine. I'm trying to figure out how the firmware determines STATE(ACCELERATOR_CALIBRATED), maybe it is possible to clear a flag in the CLI.

elgansayer, 2022-01-22:
Shame because its not just a bug in the configurator as crappy as the code for it is🙁
It happens in my own configurator tool and i see values directly streamed bad unless i do the above tricks.

cfanyu2010, 2022-01-22:
you can vhange it in cli

MrD-RC, 2022-07-21:
With the H743-WING, the accelerometer is on the underside of the flight controller. So when the flight controller is upside-down, the accelerometer is the correct way up. The accelerometer needs to be the correct way up for the first calibration step.

OptimusTi, 2022-07-21:
De de

MrD-RC, 2022-11-26:
Reset calibration button added to 6.0

closed 2022-11-26 by MrD-RC