iNavFlight/inav-configurator #1083: FC goes back into DFU after flashing firmware

undeze, 2020-10-14:
I have an Emax Hawk with a Matek F405 FC running Betaflight. I'm trying to switch to iNav. I have iNav Configurator 2.5.0 and I'm trying to flash iNav 2.5.2. My OS is Ubuntu 18.04 Linux. Each time I flash the firmware with iNav, the FC cycles back into DFU, and I'm unable to connect the FC to the configurator. I've tried various combinations of disconnecting the FC from the iNav configurator but nothing prevents the FC from cycling back to DFU. Restarting the configurator does not fix the issue. Neither does restarting my computer. With the same iNav configurator I have successfully flashed my Matek F765-WING. When using Betaflight on my F405, I have no problems. How can I prevent my F405 from cycling back into DFU after I have flashed iNav firmware? Thanks.

keatonspb, 2021-09-12:
Any ideas? Got same problem

undeze, 2021-09-12:
If I remember correctly, I gave up on iNav and stuck with Betaflight on that project. Sorry I am not on any more help. I'll check my notes later. Of course iNav is at 3.0.1 now.

avsaase, 2021-09-13:
Did you flash the correct target?

closed 2022-03-14 by DzikuVx