iNavFlight/inav-configurator #1087: The Inav Configurator 2.5.0 does not open in Windows 10

Black-Eagle954, 2020-10-22:
Inav Configurator 2.5.0 was working in my pc since it was released. Now all of sudden it does not open. I checked and tried various methods to make it work thinking probably it is a windows problem but the Configurator 2.5.0 refuses to open. Please help.

Black-Eagle954, 2020-10-22:
Betaflight Configurator 10.7.0 does not suffer from this problem, if I may say so. Many Thanks.

DzikuVx, 2020-10-22:
please define "does not open". Does it:
- not start completely?
- start minimized?
- any error throwed?
- what have you tried to fix it?
- have you tried this? https://github.com/iNavFlight/inav-configurator#inav-configurator-start-minimized-what-should-i-do

Black-Eagle954, 2020-10-24:
Hello Pawel,
It used to start minimised. But then I dug up a video by you about troubleshooting Inav. In Windows 10, I unnecessarily tried to make .exe files work.by tweaking certain parameters via the Registry Editor. But after applying your solution it worked well, I changed back the registry through my registry backed up files.
I  might add as well that before this problem, I had to disconnect my USB every time I had to configure something and reboot. Otherwise, the comport would not be detected. Now that problem has disappeared as well. I thought maybe I received a faulty Matek F722-SE FC. Now, it is working very well indeed. 
Many Thanks,
Take Care, Stay Healthy and Safe,. 

Affectionately,
Black-Eagle954

closed 2021-03-04 by DzikuVx