iNavFlight/inav-configurator #1088: OSD for SP Racing F3

Drone-pilot, 2020-10-22:
Implement support in the SP Racing F3 firmware for an external OSD based on MinimOSD, for configuration via the INAV Configurator.

DzikuVx, 2020-10-23:
We will not work on this. There is extarnal application for configuring MWOSD and similar. Both SPRF3 and minimOSD are basically end of life technologies

closed 2020-10-23 by DzikuVx

Drone-pilot, 2020-10-23:
Cleanflight and Betaflight have implemented minimosd support on spracing f3, although they also stop supporting F3 series firmware updates. Please implement, it shouldn't be difficult.

Drone-pilot, 2020-10-23:
[image]
[image]