iNavFlight/inav-configurator #1094: Add RTH Home Altitude to the configurator

MrD-RC, 2020-10-28:
For fixed wing pilots, this is a helpful feature. Especially for pilots who fly high. It would be useful to have RTH Home Altitude in the GUI.

I'm going to have a go implementing this. I'm sure I have most of it worked out, but will test tomorrow. Just leaving this here in case I can't assign it to myself.

DzikuVx, 2020-10-29:
@MrD-RC please rework it to the way it's done in #1020 with settings framework. Bear in mind you have to start from scratch as Adsvanced Tab code was just changed

MrD-RC, 2020-10-29:
So just to confirm, all you need to do now is setup the text in the locale. Then for the variable on the FC, you set the data-setting on the html input to the CLI command variable name, and it all just works. If so, awesome. That's so much simpler.

PR #1097 (2020-11-10): Add rth home altitude to advanced tuning #1094
Added RTH Home Altitude to the Advanced Tuning tab. Implemented using the settings framework.
Fixes #1094 

[image]
closed 2020-11-10 by DzikuVx