iNavFlight/inav-configurator #1114: iNav 2.6.0-RC1 Compass Calibration - within 2 degrees on 3 cardinals, wildly off on the 4th, different each time

wkiskin-teladoc, 2020-11-17:
This issue is for [https://github.com/iNavFlight/inav-configurator/releases/tag/2.6.0-RC1](url)

I have been using iNav since 2.0 and same make Matek M8Q-5883 on each build.  I can get 3 cardinal directions pretty much dead on, and the 4th one (not always same 4th cardinal) wildly off by 15 degrees.

I use a 30 foot USB extension, away from house and any metal out in the yard - same place for all previous calibrations.  No metal on my person, not even a belt buckle.

In my last attempt it was East cardinal  that was off.  When my compass indicates 89 degrees iNav configurator says 114, approx 15 degrees off.

Behavior should be like previous versions of iNav.

Jetrell, 2020-11-28:
I thought I should mention that my calibration with the 2.6 configurator was also off.  No matter how many times and places I tried to re calibrate it. Even if I calibrate the same unit at the same location, there was the inconstancy between the 2.5 and 2.6 configuratiors
It was also using a M8Q +5883.

e.g. South is always around 165°. While each time I reverted back to calibrate it with the 2.5 configurator. South shows as 180°.
I eventually just copied my old ZERO setting from 2.5 and left the GAIN'S as default.
I only noticed because I replaced a GPS on one of my quads and noticed the discrepancy in calibration. So I tried another older quad with the a 5883, and it too acted the same.

MrD-RC, 2022-11-26:
@Jetrell is this still an issue?

closed 2024-04-24 by DzikuVx