iNavFlight/inav-configurator #1115: Select 50% for Black Box logging rate, reboots and shows 25% logging rate

wkiskin-teladoc, 2020-11-18:
This issue is for [https://github.com/iNavFlight/inav-configurator/releases/tag/2.6.0-RC1](url) 

When I select 50% logging rate and click reboot, it come back at 25% logging rate.

Behavior should be it comes back at the selected rate

avsaase, 2020-11-18:
This was implemented in https://github.com/iNavFlight/inav/pull/6127 to avoid problems with logging too much data.

DzikuVx, 2020-11-19:
Like @avsaase wrote: we preven logging faster than 1kHz to prevent many issues including stopping motors and falling down from the sky

closed 2020-11-19 by DzikuVx

wkiskin-teladoc, 2020-11-19:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thank you... It was not apparent to me from using the UI.

Can you just limit the rate options in the dropdown so it "acts" appropriately?

Cheers,

Bill

CONFIDENTIALITY NOTICE: This e-mail message, including any attachments, is for the sole use of the intended recipient(s) and may contain confidential and privileged information or otherwise be protected by law. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

IgnacioHR, 2025-09-27:
Hi, Is this still an issue with new H7 processors in place?

I'm having an issue with a new 5" drone. As you can see in the picture, front engines suddenly goes up the same amount rear engines goes down! The PID reacts to correct the situation. I still don't know what is the source of the issue so I tried to record in the blackbox at higher rate to see more information and hit this "bug/feature".

iNav is running in a new H7 processor so I guess this processor could handle higher rates (I might be wrong).

I'm adding this information here just in case you think it would make sense to allow for higher rates depending on the FC processor.

[image]

I'll continue my research regarding the engines...