iNavFlight/inav-configurator #1124: Accelerometer calibration is buggy (2.6-RC1)

avsaase, 2020-11-29:
I am trying to recalibrate the accelerometer on my brand new F765-Wing. I think something went wrong wen calibrating the first time because with the flight controller flat on the table it reads 3 degrees pitch and 9 degrees roll on the Setup tab (no board alignment is set). When I go to the Calibration tab and click the button it shows the popup and greys out the six steps as expected. But when I calibrate the first axis it immediately says "Calibration finished", before it lets met calibrate the other axes. This video shows the problem: https://www.youtube.com/watch?v=4-y9DlynqHk. In this video I first tried to calibrate step 1, then 3, then 4 and then 1 again.

A second bug is that sometimes I cannot close the "Calibration finished" popup using the OK button, but I have to use cross in the upper right corner. You can see that here: https://www.youtube.com/watch?v=lJsOQU7rzhw.

Please let me know if you need more info.

cmd-line, 2021-04-01:
I am having this same issue on MacOS High Sierra with iNav 2.6.1

9Iuheni, 2021-05-16:
Same here. Inav configurtor 2.6.1, Windows 10, Omnibus F4 nano v6

petertorelli, 2021-08-16:
Having this issue on 3.0.0 and 3.0.1 on Win10 and macOS.

petertorelli, 2021-08-16:
While debugging this I notice that calibration step jumps from 1 to 6 without calling next() for any of the intermediate steps. Not sure why, but in this code `CALIBRATION_DATA.acc['Pos' + i]` is all 1's, so it skips every step. Seems to me a request to recalibate should clear all the acc.PosX values to zero. I'll mess around and see if that's the problem.

```js
    publicScope.next = function () {

        if (privateScope.step === null) {
            privateScope.step = 1;
        } else {
            var count = 0;
            for (var i = 0; i < 6; i++) {
                if (CALIBRATION_DATA.acc['Pos' + i] === 1) {
                    count++;
                }
            }

            privateScope.step = count;
        }

        console.log(privateScope.step);

        if (privateScope.step > 5) {
            privateScope.step = null;
        }

        return privateScope.step;
    };

    publicScope.getStep = function () {
        return privateScope.step;
    };
```

petertorelli, 2021-08-16:
Apparently the MSP helper [reads this from the FC](https://github.com/iNavFlight/inav-configurator/blob/b88a5004de5431bd0c461aef8913afeb21d983a6/js/msp/MSPHelper.js#L1281-L1307), and the FC says all the PosN calibrations are done, and so it ignores the GUI. So I guess something in the FC needs to be cleared in order to recalibrate?

petertorelli, 2021-09-06:
@avsaase See solution [here](https://github.com/iNavFlight/inav-configurator/issues/1070). You need to invert the craft, the click "calibrate" and it will start. Why? Dunno! But it works. A lot easier than debugging the firmware like I've been doing.

DaveBow12, 2022-10-21:
This bug still exist in INAV 5.1 and should be fixed.
It can't be that difficult to reset the parameters when someone tries to recalibrate.
Also, after the 6 position is done the software requires another click of Calibrate.  This really isn't necessary and adds confusion for first time users.  Once the 6th position is done the software should just say Done. Save Calibration.

MrD-RC, 2022-10-21:
There is already a reset calibration button in 6.0. I think we can close this now.

closed 2022-10-21 by MrD-RC

DaveBow12, 2022-10-22:
I hope it just resets the accelerometer calibration.

If so it is a reasonable solution.

Dave