iNavFlight/inav-configurator #1126: V2.6 : Flashing the board fails on MAC (sometimes)

Pairan, 2020-12-05:
## how to reproduce
* Entering the "Flash Firmware" Tab
* Select board and Version
* Load firmware online
* Flash firmware!

Usually the board should change into DFU mode and flash the picked firmware. Im my attempts (RC1 and final) the change to DFU fails and the Board is not any longer reachable. Flashing fails with "failed to open port xyz".

After two or three attempts that change to DFU succeeds and flashing is possible.

Perhaps this is something that can be pinned down and removed with the hint.

mikekenyon99, 2020-12-15:
I have tried 10+ times in a row - and I can't flash my board.

MrD-RC, 2022-11-26:
@Pairan is this still an issue with the current release? It would be good to get this closed if it is fixed.

Pairan, 2022-11-26:
Guess you're good to go, Darren ... lately I had no issues here. Otherwise I will raise a hand again.

Thanks for your support, mate!

closed 2022-11-26 by MrD-RC