iNavFlight/inav-configurator #1132: Unable to set gps_provider = MSP (2.6.0-RC1), so cannot use the M9N-CAN GPS unit at this time

wkiskin-teladoc, 2020-12-09:
# version
# INAV/MATEKF722 2.6.0 Nov  6 2020 / 14:03:17 (d0046f30)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# set gps_provider = MSP
### ERROR: Invalid value. Allowed values: NMEA, UBLOX, UNUSED, NAZA, UBLOX7, MTK

Despite what I am seeing in Pawel's video:

[https://youtu.be/0hN1EJqmjUk](url)

 "MSP" is an illegal value for gps provider and so the new Matek cannot be used withe 2.6.0-RC1

Expected Behavior: I should be able to use the Matek M9N-CAN GPS unit as I see in the YouTube video from Pawel on iNav 2.6.0-RC1

giacomo892, 2020-12-09:
I've tested it on F765-WING using 2.6 and it is working just fine

digitalentity, 2020-12-10:
Please try stable release.

DzikuVx, 2020-12-10:
2.6.0-RC1 was released on Nov 15th. Your firmware is from Nov 6th. Update to 2.6 final  version

closed 2020-12-10 by DzikuVx