iNavFlight/inav-configurator #1148: Automatic backups

skorokithakis, 2020-12-29:
It would be great if we could specify a directory in the options, and have the configurator do a "diff all" and write the settings in a file in that directory every time we saved.

Basically, this would automate the "cli -> `diff all` -> save to file" workflow and allow for constant backups, because lots of friends just had the FC reset settings on them suddenly.

gdanas87, 2020-12-29:
@skorokithakis agreed! Thank you for the suggestion!!!

skorokithakis, 2021-09-13:
Any movement on this? It seems like a relatively easy feature to implement (all that needs to happen is to write the initial loaded settings to a file) and would save many people headaches.

DzikuVx, 2021-09-14:
You can do it yourself if you have some spare time

StuweFPV, 2022-01-10:
I don't know how you suddenly have FC reset settings (assuming you mean defaults after a flash) in your FC.. you would really have to copy the wrong file. there is the date at the top of each file so how can that happen?
I work on a plane by plane basis and usually work on one plane at one. whenever i do a dump and file it, it opens up the last used directory - the plane i'm working on and that's perfect. i would not want all my inav dump files in one single directory.. what a mess. it's also wise if you rename the files prior to saving with a version number so you know which one you were working on last. personally i have no need to change the way it is.

skorokithakis, 2022-01-10:
If you have no need, this ticket isn't about you. Just ignore it and move on.

closed 2022-01-10 by skorokithakis

StuweFPV, 2022-01-10:
well sorry about that.. just wanted you to show you another way to solve your problem.. but please accept my apologies for trying to help... i'll move on

skorokithakis, 2022-01-10:
I posted a comment by accident, I meant to just close the ticket instead. I've removed it now.