iNavFlight/inav-configurator #1154: [Feature Request] add POI, Head, Jump to Mission planner tab

ArnoTlse, 2021-01-04:
Hi,

I don't really know where to put my message, so I put it here but maybe there will be a better place then.

I'm quite recent in the iNav adventure as I have restarted playing with drones in begining of 2020 after a 10 years break after leaving engineering school. Since a lot has evolved. Thus, I have had to make a whole update of my rusty hardware and software alike, and I have discovered many new software/firmware platform such Betaflight, Ardupilot and iNav.

I have since 8 months chosen iNav as it fits well my needs to fly a quad with or without GPS navigation mode alike. I'm impressed of all the work done and the regular updates. I'm not a professional coder, but I know some python/JS/html stuffs for my job/leisure needs, and when I have seen in the 2.6 released that there were the ability to select Point of Interest or Heading Lock for navigation purposes in iNav Flight controller firmware, I wonder if it could be possible to plan such a mission directly in iNav configurator, as I was able to select only Waipoints. In parallel, I have tested also MWPtools installed on a jetson nano that works well. However, it is not convenient for me to bring the nano + a display + a PSU on the airfield, as I have no other Linux computer. Therefore, I found that it could be a good exercice to go deeper in the iNav configurator code and see if I manage to do something on the mission planer tab.

Today, work is in progress and I have reached something that allow me to plan a mission directly in the configurator tab, taking into account PoI, set_HEAD, JUMP and classic WP & Position Hold. I am also able to generate the XML mission file like the one generated by MWPtools.

[image]

I'm now tackling the "send mission to FC" part through MSPcode.

I wonder if anyone is also already working on the same thing for a futur iNav Configurator released ? Or if this kind of stuff could have an interest in the iNav community ?
I used to have a tablet PC and a Android smartphone with me on airfield. I used the Tablet PC for iNav configurator linked to the Quad, and the smartphone linked to my Xlite with "telemetry viewer" installed on it. That's why I find convenient to plan a mission directly though the configurator.
I wonder also if it is something feasible to send (uplink) a mission from the iNav Configurator to FC through Sport/Fport receiver & OpenTX radio (such the xlite with built-in Bluetooth) ? Or if it will be mandatory to go through another transmission device such Lora ESP board directly connected to the FC ?

Regards & best wishes for iNav in 2021.

Arnaud

artemsh, 2021-01-05:
Hi, looks great, hopefully coming in 2.7!

ArnoTlse, 2021-01-10:
Hi, 

Thanks. So I go on working on it then. 
I have tested several scenarios to test first XML output mission.
I am working on the MSP code to write the mission inside the FC, but I have to understand a little bit more how MSP works.

**JUMP scenario**

[image]

```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<mission>
	<version value="2.3-pre8"/>
	<mwp cx="-2.0803485" cy="48.3578699" zoom="16.22428040534076"/>
	<missionitem no="1" action="WAYPOINT" lon="-2.0852908" lat="48.3600744" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="2" action="WAYPOINT" lon="-2.0836285" lat="48.3575967" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="3" action="WAYPOINT" lon="-2.0806069" lat="48.3561077" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="4" action="WAYPOINT" lon="-2.0763731" lat="48.3572367" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="5" action="WAYPOINT" lon="-2.079762" lat="48.3595861" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="6" action="JUMP" lon="0" lat="0" alt="0" parameter1="2" parameter2="1" parameter3="0"/>
</mission>
```

**POI + LDG scenario**

[image]

```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<mission>
	<version value="2.3-pre8"/>
	<mwp cx="-2.0811254" cy="48.3582818" zoom="16.491037275330594"/>
	<missionitem no="1" action="WAYPOINT" lon="-2.0853464" lat="48.3600205" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="2" action="WAYPOINT" lon="-2.0832547" lat="48.3574387" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="3" action="SET_POI" lon="-2.0801173" lat="48.357601" alt="50" parameter1="0" parameter2="0" parameter3="0"/>
	<missionitem no="4" action="WAYPOINT" lon="-2.0812929" lat="48.3561452" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="5" action="WAYPOINT" lon="-2.0761936" lat="48.3575756" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="6" action="WAYPOINT" lon="-2.0801478" lat="48.3600611" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="7" action="SET_HEAD" lon="0" lat="0" alt="0" parameter1="-1" parameter2="0" parameter3="0"/>
	<missionitem no="8" action="LAND" lon="-2.0847357" lat="48.3601726" alt="50" parameter1="0" parameter2="0" parameter3="0"/>
</mission>
```

**SET_HEAD + RTH scenario**

[image]

```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<mission>
	<version value="2.3-pre8"/>
	<mwp cx="-2.0805753" cy="48.3587484" zoom="16.242279953042264"/>
	<missionitem no="1" action="WAYPOINT" lon="-2.0853464" lat="48.3600205" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="2" action="WAYPOINT" lon="-2.084667" lat="48.3590923" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="3" action="SET_HEAD" lon="0" lat="0" alt="0" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="4" action="WAYPOINT" lon="-2.0792165" lat="48.3591684" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="5" action="WAYPOINT" lon="-2.0784608" lat="48.3585495" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="6" action="WAYPOINT" lon="-2.0843845" lat="48.3584176" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="7" action="WAYPOINT" lon="-2.0834456" lat="48.3575503" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="8" action="WAYPOINT" lon="-2.0770028" lat="48.3578343" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="9" action="WAYPOINT" lon="-2.0762241" lat="48.3571698" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="10" action="WAYPOINT" lon="-2.0824532" lat="48.3567539" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="11" action="SET_HEAD" lon="0" lat="0" alt="0" parameter1="-1" parameter2="0" parameter3="0"/>
	<missionitem no="12" action="WAYPOINT" lon="-2.0856122" lat="48.3578793" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="13" action="RTH" lon="0" lat="0" alt="50" parameter1="1" parameter2="0" parameter3="0"/>
</mission>
```

**2 POI + Position Hold Timer scenario**
[... 10 lines cut ...]
	<missionitem no="3" action="SET_POI" lon="-2.082751" lat="48.3596534" alt="50" parameter1="0" parameter2="0" parameter3="0"/>
	<missionitem no="4" action="WAYPOINT" lon="-2.0808143" lat="48.3600142" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="5" action="WAYPOINT" lon="-2.0832088" lat="48.3607555" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="6" action="WAYPOINT" lon="-2.0844333" lat="48.3599539" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="7" action="WAYPOINT" lon="-2.0835081" lat="48.3590077" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="8" action="SET_HEAD" lon="0" lat="0" alt="0" parameter1="-1" parameter2="0" parameter3="0"/>
	<missionitem no="9" action="WAYPOINT" lon="-2.0786646" lat="48.3567597" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
	<missionitem no="10" action="SET_POI" lon="-2.0787962" lat="48.3581093" alt="50" parameter1="0" parameter2="0" parameter3="0"/>
[... cut ...]
	<missionitem no="17" action="PH_TIME" lon="-2.0816684" lat="48.3569199" alt="50" parameter1="150" parameter2="0" parameter3="0"/>
</mission>
```

That's it for the GUI part.

If anyone can help me, I have a question so far on the Altitude units. In the configurator, the Altitude units is defined in the input field in cm (with a label converting it in m for information). But, in the XML file, it is saved in m, not in cm. On FC point of view, does the FC wait for altitude in cm or in m ?

Regards

Arnaud

ArnoTlse, 2021-01-10:
I have also in mind some other stuff for future steps if it can be considered useful:

- [ ] ability to insert point between 2 existing WPs.
- [ ] ability to draw patterns such circular or rectangular with interactive parametrics (such number of WP or radius/length drag selectors)

Regards

ArnoTlse, 2021-01-23:
Hi,

Can someone help me understanding how MSP.send_message works ?

I'm stuck on the latest part of adding POI,HEAD, Jump, i.e on the LoadFromFC/EEPROM. I am able to write the mission on it but not decode it reversely in the configurator.

In the function getNextPoint() in Mission_control.js, the function is iterating on itself I guess through this callable function :

`MSP.send_message(MSPCodes.MSP_WP, mspHelper.crunch(MSPCodes.MSP_WP), false, getNextPoint);`

until the counter of points is reached.

However, as I manage Set_HEAD, JUMP and RTH as an option of Physcial WP (WAYPOINT, SET_POI, PH, PH_TIME, LAND), I have to store the history of all the point in the buffer. At least, the last buffer state.

Is there a possibility to pass a function with arguments to the MSP.send_message() functions where the argument would be the previous buffer state ?
Or is there an options to call all the buffer points in once ?

Regards

Arnaud

egibert1996, 2021-02-07:
Hi there,
Can you please explain me what exactly does the JUMP action do, I don't understand the difference between the WAYPOINT action that goes from one point on the map to another an JUMP action. And what means repeat parameter? 

Thanks

ArnoTlse, 2021-02-11:
> Hi there,
> Can you please explain me what exactly does the JUMP action do, I don't understand the difference between the WAYPOINT action that goes from one point on the map to another an JUMP action. And what means repeat parameter?
> 
> Thanks

Hi, 

From what I have understood so far, because I am not the developer of this Waypoint kind in Inav firmware, I have just developed the Interface to play with it in Inav configurator, the JUMP action enables to loop over some part of your mission without having to recreate those points.
If you take my first example (called Jump Scenario), once you reach the Waypoint 5, as I have defined a Jump action to Waypoint 2 with 1 iteration loop (information in pink on the dash line), your Quad will go to Waypoint 2 and fly again one time WP2 to WP5. 
Thus, it allows you to loop over some existing points without recreating several identical WP; it can be useful coupled with a Point-of-Interest WP for aerial photography flight as instance.

I don't know whether it is clearer for you now ?

regards

Arno

ArnoTlse, 2021-02-15:
Hi,

I manage to find out how to make things work with MSP. It was not easy when you're not get used to it. 
I am then able to save and load from FC mission with all kind of waypoint (set_head, POI, JUMP, RTH, LAND...).
I have just to finish debugging something on Eprom side and I will be able to launch a PR whether someone wants to check if I am not doing something wrong.

Regards

ArnoTlse, 2021-02-27:
This feature has been developed and PR Is currently under review. I close this issue so.

closed 2021-02-27 by ArnoTlse

supernova1969a, 2022-07-05:
I'm not sure how to do this, the set_head command maybe. I'd like to while Pos Holding spin the quad around while holding, maybe pausing at each heading. I'm not sure I understand set_head to start with if someone could explain it, otherwise I think this would be great if its possible. Thanks!!

stronnag, 2022-07-05:
> I'm not sure I understand set_head to start with if someone could explain it, otherwise I think this would be great if its possible. 

Could always try the [wiki description](https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages#set-poi-multirotor-only-multiwii-inav-26-and-later)

supernova1969a, 2022-07-05:
Thank you. I've been looking for this! I apologize if I came off stupid..

supernova1969a, 2022-07-05:
So after reading the wiki. I'm not clear as to how I would do a hold_and_spin..while hold at a waypoint. And how you would tell it to Look at that direction for a length of time. If this feature does not exist I think it would be a great addition. Again at a waypoint while holding for t time, can I set_head x for t, repeat y times at same way point.