iNavFlight/inav-configurator #1159: Presets in configurator 2.6 use CLI variables which were deleted for this release

grizewald, 2021-01-18:
Some presets defined in preset_definitions.js still use CLI variables which were deleted in the 2.6.0 release. For example, gyro_notch1_hz is used in the 3" Cinewhoop preset. 
None of the presets have values for the new Control Derivative gain. Is it really correct that no differences exist in this setting for any size multirotor craft?
Also, the global FILTER_CONFIG settings have entries for gyroNotchHz1 and gyroNotchHz2 when there is only one such setting now.

It would be really nice to have some presets which reflect the way the PID controller is tuned in the latest release.

Phil-MC, 2021-02-05:
I agree with grizewald. I was setting up a 3" Cinewhoop and assumed the presets would have good values for the current release. Now I am wondering what else is suspect!

grizewald, 2021-02-22:
It's a real shame that there is no response to this.

If I had a small army of multirotors and didn't have to work to pay my bills, there's nothing I'd more like to do than spend all my days flying and tuning, but I'm nowhere near as close to retirement as I'd like to be.

In the end, the presets become a liability if nobody is prepared to update them each time a major release completely changes tuning and PID parameters. If nobody has the time to keep the presets up to date, wouldn't it be a better idea just to remove them completely?

DzikuVx, 2021-08-13:
made obsolete #1327

closed 2021-08-13 by DzikuVx