iNavFlight/inav-configurator #1164: Motor mixer values are shown incorrectly

avsaase, 2021-01-21:
When using a custom motor mixer with decimal values, such as 
```
# mmix
mmix 0  1.000 -0.887  0.723  1.000
mmix 1  1.000 -1.000 -0.723 -1.000
mmix 2  1.000  0.887  0.723 -1.000
mmix 3  1.000  1.000 -0.723  1.000
```
then the configurator shows these value incorrectly:
[image]

This happens with both the current master branch and release 2.6.1 for Windows 10.

DzikuVx, 2021-01-21:
It happens on all the releases and is caused by the fact that MSP layer is integer not float based

avsaase, 2021-01-21:
Wouldn't it be possible to just round it to three decimals in the configurator? Since the difference to the correct value is so small this should always lead to the same value as the CLI command.

DzikuVx, 2022-03-14:
This was fixed some time ago so closing

closed 2022-03-14 by DzikuVx