iNavFlight/inav-configurator #1169: Feature request: Tooltip on PID tuning page showing the name of the setting

Sebbb, 2021-02-05:
It's hard to identify which setting on the PID tuning page maps to which setting inside the configuration. A tooltip shown when hovering the mouse over the field, showing that name would be helpful

StuweFPV, 2022-01-10:
or just a consistent naming convention.. when you go to the filters it's even worse. one name for the gui, one for the code and one in the wiki. all different but one and the same value. it's hard to keep it under control when so many different people work on the same thing. that's just how it is.. no biggie.
i found the best way to identify what you want is. do a dump, go into the configurator and change only that value you want, do another dump and compare the two files with Winmerge or something similar. it pops right out and you got it. search for the code in the wiki then you have the correct description.

MrD-RC, 2022-11-26:
Tool tips are on most settings now, everywhere. Clicking the question mark will also take you to the appropriate place in the settings document.

closed 2022-11-26 by MrD-RC