iNavFlight/inav-configurator #1177: Inav Configurator 2.6.1 unstable on M1 Mac - Big Sur

EmanuelStassar, 2021-02-23:
I am running Inav Configurator on an M1 Mac and its not very stable. Sometimes it crashes on startup, sometimes after a few minutes. It's borderline usable this way. 

I hope this app can be converted into a universal app for Mac so it can run natively on M1 processors.

WillsGledhill, 2021-02-23:
Same for me on big sur but not M1 ive got the 2020 intel 13" pro

bangkokopete, 2021-04-11:
not only on M1 , also here on Macbook 13" 2016 Big Sur. crashes always on start up , runs after the second or third attempt and crashes in the middle of the usage. My feeling only has something to do with com port access. 
BTW does happen on Betaflight configurator on start up as well, but only occasionally.

mikekenyon99, 2021-04-17:
Hi - same here.  Apple M1 Mac Mini.  Pretty much useless, crashes after it has connected every time.

al3k87, 2021-04-20:
This is a Big Sur issue with node.js, betaflight had same issue also but they have updated the version so its now stable. please fix this, is there any access to nightly builds that possible allready have this fix?

PR #1238 (2021-04-21): Bump NW.js to 0.52.3
hopefully Closes #1177
closed 2021-04-21 by DzikuVx