iNavFlight/inav-configurator #1181: [Mission Planner] new features development

ArnoTlse, 2021-02-27:
Hi,

I am currently planning working on those topics related to Mission planner tab in Inav Configurator : 

- [ ] **Safe Home** : add Safe Home settings menu to manage and plot safe home point accessible through a similar button on the map as settings
- [ ] **WP insertion** : add ability to insert a waypoint in existing mission by clicking on the branch where user wants to insert new point
- [ ] **Mission pattern template** : add mission pattern template such circular, rectangular, scanning with options to modify size and number of WP
- [ ] **mission play** : ability to play a mission, i.e having a vehicle icon moving on the mission taking into account POI, SET HEAD, JUMP
- [ ] **mission planner telemetry** : ability to plot live or differed telemetry (GPS position, alt, speed and heading at least ) on mission planner. For live telemetry, I would like to be able to extract telemetry data of my taranis xlite through BT and send it to Inav configurator mission planner tab. Hope this is feasible in Inav Configurator? I know thanks to @DzikuVx video we can read live telemetry Data from opentx radio to a telemetry app.

I will update this issue each time a feature is developed. 

Arno

MrD-RC, 2021-03-30:
Awesome. All are great, but the safehomes setting would be very useful, even for people who don't fly missions.

Mateyhv, 2021-04-19:
Fantastic!! All those updates will be awesome. Thanks Arno!

avsaase, 2021-04-28:
Is it me or is there no button to save the waypoint altitude and speed anymore? Iirc there used to be a button for this, but now I can only save the changes by using the return key. Was this changed?

ArnoTlse, 2021-04-28:
> Is it me or is there no button to save the waypoint altitude and speed anymore? Iirc there used to be a button for this, but now I can only save the changes by using the return key. Was this changed?

Hi,

yes, I've removed the save button to simplify the integration of Optional Non geo-point, as adding/removing/modifying the non geo-point table is automatically saved now.  Thus a Save button would have done the action twice. 

Arno

avsaase, 2021-04-28:
Okay, that makes sense, but can you also make the changes to the WP altitude and speed save automatically? It doesn't say anywhere that hitting return saves the WP, I only found out accidentally. There is also no feedback that it actually saved. This would be fine if it always saved immediately.

breadoven, 2021-04-28:
Just had a play with the new Mission Control which is looking great. A few things I noted in the latest version were :

1) It's possible to set an intermediary WP as RTH which isn't possible any more, it can only be the last WP.
2) RTH landing is set by changing the P1 value. Is there an intention to make this more intuitive than knowing to set P1 = 1 ?
3) Map doesn't focus in on a newly loaded mission leaving you having to zoom out to find it
4) Would be good if the loaded filename was displayed somewhere

Still a work in progress so I'm sure most of the above are just pending finding the time.

Just need to get absolute (true) altitude datums (https://github.com/iNavFlight/inav/pull/6662) added into the mix and INAV missions will be highly configurable.

ArnoTlse, 2021-05-04:
> Okay, that makes sense, but can you also make the changes to the WP altitude and speed save automatically? It doesn't say anywhere that hitting return saves the WP, I only found out accidentally. There is also no feedback that it actually saved. This would be fine if it always saved immediately.

Hi,

Yes sure, I can change it in this way. I'll put it in my todo list in the next PR.

Arno

ArnoTlse, 2021-05-04:
> Just had a play with the new Mission Control which is looking great. A few things I noted in the latest version were :
> 
> 1. It's possible to set an intermediary WP as RTH which isn't possible any more, it can only be the last WP.
> 2. RTH landing is set by changing the P1 value. Is there an intention to make this more intuitive than knowing to set P1 = 1 ?
> 3. Map doesn't focus in on a newly loaded mission leaving you having to zoom out to find it
> 4. Would be good if the loaded filename was displayed somewhere
> 
> Still a work in progress so I'm sure most of the above are just pending finding the time.
> 
> Just need to get absolute (true) altitude datums ([iNavFlight/inav#6662](https://github.com/iNavFlight/inav/pull/6662)) added into the mix and INAV missions will be highly configurable.
> 
> Edit: forgot about support for previous versions of INAV which do allow multiple RTHs in a single mission. So scrub comment 1) above.

Hi @breadoven 

Thanks for the feedback. I'll take note about the points you mention. I try to have a look on it for the next PR.
For the focus in issue, when the pb occurs? It is when you have already a mission loaded and you want to reload another one or at the first loading? Is it from mwp files or from FC/EPROM loading or both?

Arno

ArnoTlse, 2021-05-06:
hi @breadoven ,

On the third point you mention on Map not focusing, I'm not able to reproduce it. When I load a new file, the Map focuses automatically to the first WP. Can you provide me more information on your issue ?

Arno

breadoven, 2021-05-06:
> hi @breadoven ,
> 
> On the third point you mention on Map not focusing, I'm not able to reproduce it. When I load a new file, the Map focuses automatically to the first WP. Can you provide me more information on your issue ?
> 
> Arno

You're right. Loading a mission file works OK. It's actually loading a mission from EEPROM that doesn't auto focus.

ArnoTlse, 2021-05-06:
Ok, I have found the issue for EPROM map focus. I have correct it. I will push it with the other things (filename display, WP speed & Alt on the fly change, Distance computation with Jump).

Arno

ArnoTlse, 2021-05-08:
Hi @breadoven ,

on the [iNavFlight/inav#6662 ](https://github.com/iNavFlight/inav/pull/6662) topic, just to be sure, the P3 parameter to be set for indicating reference from Take-off level or Sea-level is only intended to be used for Non-Geo point, ie Waypoint, Land & PositionHold ?

Arno

ArnoTlse, 2021-05-08:
I have added the P3 interface for Altitude reference like that : 

[image]

ArnoTlse, 2021-05-08:
> Correct Arno, the absolute datum is just for geospatial WPs, so Waypoint, Land & PositionHold, i.e. those with an altitude setting. (I assume you meant geospatial rather than Non Geo ?).
> 
> Is there any intention of pulling in ground elevation heights from the web to be shown for each WP ? Would be useful in the long run.

Ok, thanks.
I have added your point for "elevation" in the ToDo list. It could be interesting to display it as well, sure. I will figure out how to do it.

Arno

breadoven, 2021-05-12:
> Ok, thanks.
> I have added your point for "elevation" in the ToDo list. It could be interesting to display it as well, sure. I will figure out how to do it.
>
> Arno

I've been using https://www.opentopodata.org/ Arno. Don't know if this is suitable for INAV or not. I'm sure there are better sources of DEM data.

stronnag, 2021-05-12:
> I've been using https://www.opentopodata.org/ Arno. Don't know if this is suitable for INAV or not. I'm sure there are better sources of DEM data.

FYI, [mwp](https://github.com/stronnag/mwptools) is using [Bing elevations](https://docs.microsoft.com/en-us/bingmaps/rest-services/elevations/) for its inav 3.0 [absolute <=> relative / landing altitude /  terrain analysis / clearance](https://github.com/stronnag/mwptools/wiki/Support-for-inav-3.0-WP-features) features. Whilst it's a closed source, it provides excellent global coverage with the usage / performance issues that some of more open DEM provides have. In any case, to use the configurator's mission planning, Bing is pretty much a requirement anyway (same API key for tiles and elevations).

rts18, 2021-05-13:
@ArnoTlse I noticed that the `Wait time` isn't displayed unless the mission data is entered via the CLI.
Would you consider adding it to this interface; for more simplicity of use?

[image]

ArnoTlse, 2021-05-13:
> FYI, [mwp](https://github.com/stronnag/mwptools) is using [Bing elevations](https://docs.microsoft.com/en-us/bingmaps/rest-services/elevations/) for its inav 3.0 [absolute <=> relative / landing [...]

Hi @breadoven, hi @stronnag,

OK, thanks for the feedback. I have a quick look at both sources. Bing is already usable through the map Proxy features, so I guess using elevation from it should be done adapting the existing code for Bing api. I have not use already Bing, but it is the opportunity to do so.
However, I wonder also if Google map api could also be used?

ArnoTlse, 2021-05-13:
> @ArnoTlse I noticed that the `Wait time` isn't displayed unless the mission data is entered via the CLI.
> Would you consider adding it to this interface; for more simplicity of use?
> 
> [image]

Hi @rts18,

Thanks for the feedback. In the version I have in dev for the next PR coming soon (I try to finish the algorithm to compute mission distance taking into account Jump), this is already corrected.

Arno

stronnag, 2021-05-13:
> However, I wonder also if Google map api could also be used?

The Google licensing is rather more problematic compared to the other sources. Bing is easy; simple JSON api, no real world usage restrictions. Beats me why there is not just a product key rather than forcing the user to get one; a number of open source tools have embedded Bing API keys, with, I would anticipate many more users that the Configurator.

ArnoTlse, 2021-05-13:
> The Google licensing is rather more problematic compared to the other sources. Bing is easy; simple JSON api, no real world usage restrictions. Beats me why there is not just a product key rather [...]

OK, I will look first at Bing solution as it is already in place for the tiles. 
I see that it uses Rest API to request the elevation using lon, lat coordinate un WGS84 Referential (ellipsoid). So, it seems I have to write a send/receive request function and extract the info from the receive JSON to display elevation on the GUI. Just to confirm, in Inav Firmware, the altitude datum is intended to be reference to Geoid SEA level reference not the ellipsoid WGS84? 

I have also to figure out the resolution, to see if I can display a plot of the elevation under the mission path at a certain sampling and not only for each waypoint. 

Another question on a different topic, according to jump repeat : is a repeat equal to 0 is equivalent to have no jump in fact, or does it mean that the jump is done once only?

stronnag, 2021-05-13:
> I see that it uses Rest API to request the elevation using lon, lat coordinate un WGS84 Referential (ellipsoid). So, it seems I have to write a send/receive request function and extract the info [...]

In inav, as all altitudes are ultimately from the GPS, hence they are WGS84 ellipsoid.
 
> I have also to figure out the resolution, to see if I can display a plot of the elevation under the mission path at a certain sampling and not only for each waypoint.

Depends on location,but typically 30m, Bing allows you 1024 locations in a single request. [mwp](https://github.com/stronnag/mwptools) terrain analysis requests the lesser of 1024 and "distance_metres"/30, where "distance_metres" ignores jumps because its too confusing to view on an elevation plot (for me at least).
 
> Another question on a different topic, according to jump repeat : is a repeat equal to 0 is equivalent to have no jump in fact, or does it mean that the jump is done once only?

No jump.

ArnoTlse, 2021-05-13:
OK thanks. 
So it is worth I ask it. From [this inav PR](https://github.com/iNavFlight/inav/pull/6662) , I have understood that It was Sea level reference (geoid) and not WGS84 (ellipsoid). Thus, it was a misunderstanding from my side. I'll request the elevation in WGS84 so.

I wonder on the jump case  for elevation, if you do not take them into account to avoid displaying similar branch several times, how do you plot the elevation on the jump branch to rally the jump target? This branch can be useful to display at least no?

stronnag, 2021-05-13:
> OK thanks.
> So it is worth I ask it. From [this inav PR](https://github.com/iNavFlight/inav/pull/6662) , I have understood that It was Sea level reference (geoid) and not WGS84 (ellipsoid). Thus, it was a misunderstanding from my side. I'll request the elevation in WGS84 so.

Just terminology I think; it's impossible for inav to know the local geoid MSL.

> I wonder on the jump case for elevation, if you do not take them into account to avoid displaying similar branch several times, how do you plot the elevation on the jump branch to rally the jump target? This branch can be useful to display at least no?

It would be useful, perhaps confusing and I'll have needed to do much more work so I only display each jump once. Sure, possible, maybe one day, but for now a  "ToDo".

Mateyhv, 2021-05-18:
@ArnoTlse first of all, WOW man! You are doing an amazing job updating and upgrading the mission control!
I have recently played with it and have a couple of suggestions over what is already done.

- would be nice to have different lines showed on the map for backward JUMPs (wp5 to wp3) opposed to forward JUMPs (wp3 to wp5) eg: the current magenta dashed line for one direction and double dashed line for the opposite direction. This came up on a recent quite convoluted mission I did where later on I had a hard time to find the forward JUMP when all jumps had the same line representation.
- on the Edit point window would be useful to have the number of the point, eg "Edit point 8"

Thanks for the hard work! :-)

ArnoTlse, 2021-05-19:
Hi,

here is a first draw of the Elevation at WP level extracted from Bing Map API.
The elevation is updated once the WP Sea Level ref is toggled and when the WP is moved (update only done when mouse action is up to avoid requesting to many times BING Elevation as there is a limitation per year of Bing request).

[image]

I am now working on the Elevation profile + mission altitude profile plot underneath the Map. Elevation plot will be available by similar button  as default settings. Dedicated icon will come.

Arno

ArnoTlse, 2021-05-19:
> @ArnoTlse first of all, WOW man! You are doing an amazing job updating and upgrading the mission control!
> I have recently played with it and have a couple of suggestions over what is already done.
> 
> * would be nice to have different lines showed on the map for backward JUMPs (wp5 to wp3) opposed to forward JUMPs (wp3 to wp5) eg: the current magenta dashed line for one direction and double dashed line for the opposite direction. This came up on a recent quite convoluted test mission I did where later on I had a hard time to find the forward JUMP when all jumps had the same line representation.
> * on the Edit point window would be useful to have the number of the point being currently edited, eg: "Edit point 8"
> 
> Thanks for the hard work! :-)

Hi @Mateyhv ,

Thanks a lot for your feedback. It has been also possible thanks to the expertise and advises from Stronnag.
I take the point on your suggestions. It is something I can put in the next PR I think.

Arno

ArnoTlse, 2021-05-26:
Hi, 

Here will be the differentiation for Forward vs backward Jump. I have put some arrows to indicate direction of Jump.

[image]

[image]

Arno

Mateyhv, 2021-05-27:
Thanks @ArnoTlse ! Much easier to understand whats going on :-)

ArnoTlse, 2021-05-29:
Hi,

Elevation profile is in progress : 

- Elevation from Bing is extracted now and ploted
- Mission elevation profile is also extracted and ploted

[image]

Still work in progress.

Arno

ArnoTlse, 2021-05-30:
Hi,

I have added a Home point to define Take Off location in order to get the WGS84 reference for Mission elevation. When WP altitude reference is set to WGS84 instead of Take Off location, the elevation of the WP is updated according to its reference.

[image]

Some cleaning work remaining prior to a new PR soon.

Arno

breadoven, 2021-05-30:
Shouldn't this be the other way around Arno, i.e. if the WP altitude is relative to takeoff then the WP elevation is corrected to the WGS84 elevation (SL reference) based on the home elevation in order that it displays correctly in the elevation plot ? WPs that use the SL ref need to be set based on the desired absolute (true altitude in aviation speak) altitude only, the takeoff (home) elevation isn't relevant. It's only relevant to WP altitudes set relative to the takeoff. It's a bit tricky because you can only assume where the takeoff is likely to be but I suppose there's nothing you can do about that, just be aware that WPs set with a relative altitude will see that altitude vary in absolute terms if the takeoff position and therefore also the takeoff elevation changes.

stronnag, 2021-06-03:
@ArnoTlse  FYI, I recently added a planning assumed home to the optional / "private" `mwp` stanza section of the XML file that you might find convenient to use as well (as you're already using `cx,cy`). [Reference XSD]( https://github.com/stronnag/mwptools/blob/master/src/samples/mw-mission.xsd).

ArnoTlse, 2021-06-03:
Hi,

I'm not sure If I have understood well your point.

> Shouldn't this be the other way around Arno, i.e. if the WP altitude is relative to takeoff then the WP elevation is corrected to the WGS84 elevation (SL reference) based on the home elevation in [...]

What I have done here is :

- loading WGS84 elevation (Sea level reference) under the mission line to plot the earth profile beneath the lines. this is the orange area-line.
- retrieving WPs altitude for each point and plot them regarding : 
         - if Alt reference is set to Take-Off location (P3=0), I plot the sum of WGS84 elevation beneath the HOME (Purple Icon) positionned by the user (i.e the Take-Off elevation reference) with the WP altitude enter by the user
         - if Alt reference is set to Sea Level (WGS84 ellipsoid reference), I plot directly the WP altitude value entered by the user.

> It's a bit tricky because you can only assume where the takeoff is likely to be but I suppose there's nothing you can do about that, just be aware that WPs set with a relative altitude will see that [...]

Indeed, that's why I have had to add a Home (Take-off) location icon to let user assume where will be its Take-off location to ensure that height (when Take-off reference related) will not interfere with Earth elevation extracted from Bing Database. When the user moves the Home icon, it automatically updates the elevation plot and move the blue curve (for Take-off reference WP) accordingly.

Arno

ArnoTlse, 2021-06-03:
> @ArnoTlse FYI, I recently added a planning assumed home to the optional / "private" `mwp` stanza section of the XML file that you might find convenient to use as well (as you're already using `cx,cy`). [Reference XSD](https://github.com/stronnag/mwptools/blob/master/src/samples/mw-mission.xsd).

Hi @stronnag ,

Ok thanks for the information.
I guess it is the two tags home-x / home-y ?
I will look at connecting those fields to the GUI in order to load/save HOME from/to a mission file.

Arno

stronnag, 2021-06-03:
> Hi @stronnag ,
>
> Ok thanks for the information.
> I guess it is the two tags home-x / home-y ?
> I will look at connecting those fields to the GUI in order to load/save HOME from/to a mission file.
>
> Arno

That's correct. Great if we both use the same attribute tags.

avsaase, 2021-06-04:
@ArnoTlse I noticed I cannot create waypoints when I don't have a flight controller connected. Is that WAI?

stronnag, 2021-06-04:
> @ArnoTlse I noticed I cannot create waypoints when I don't have a flight controller connected. Is that WAI?

No, works fine here.

avsaase, 2021-06-04:
Nope, really does not work here. Step to reproduce:

1. Open Configurator
2. Go to Mission Control
2. Try to create WPs: doesn't work
3. Connect FC
4. Disconnect FC
5. Go back to Mission Control
6. Try to create WPs: now it does work

This is with a fresh build of the `release_3.0.0` branch. Someone on facebook reported the same problem with RC2.

stronnag, 2021-06-04:
Whatever, still works just fine here, without an FC in the room.

ArnoTlse, 2021-06-10:
> That's correct. Great if we both use the same attribute tags.

Hi @stronnag ,

I have implemented the load and save home-x / home-y for home in a mission file. Will be in the next PR with Elevation.

Arno

ArnoTlse, 2021-07-08:
Update of the on going list to add #1311 topics

breadoven, 2021-10-17:
Don't know if you're still working on this Arno but thought I should mention I've got a working solution for Multi Missions (https://github.com/iNavFlight/inav/pull/6765). Have you done anything related to this ?

ArnoTlse, 2022-03-01:
Hi, 

Some updates here : https://github.com/iNavFlight/inav-configurator/issues/1311#issuecomment-1055820004

Arno