iNavFlight/inav-configurator #1182: [Need Developers Tips&Help] to understand how to implement MSPCodes.MSP2_INAV_SAFEHOME

ArnoTlse, 2021-02-28:
Hi,

I am working on implementing SAFEHOME management in Inav Configurator Mission planner tab.

I have modified the following file :

- MSPhelper.js  by adding a case for MSPCodes.MSP2_INAV_SAFEHOME:
```
if (!dataHandler.unsupported) switch (dataHandler.code) {
    case MSPCodes.MSP2_INAV_SAFEHOME:
          console.log("toto");
          SAFEHOME.number = data.getUint8(0);
          SAFEHOME.enable = data.getUint8(1);
          SAFEHOME.lon = data.getInt32(2);
          SAFEHOME.lat = data.getInt32(3);
          break;
    default:
          console.log('Unknown code detected: ' + dataHandler.code);
} else {
    console.log('FC reports unsupported message error: ' + dataHandler.code);
}
```
- MSPcodes.js by adding the 2 MSP2_INAV SAFEHOME command 
```
    MSP2_INAV_OPFLOW_CALIBRATION:       0x2032,
    
    MSP2_INAV_SAFEHOME:                 0x2038,
    MSP2_INAV_SET_SAFEHOME:             0x2039
};
```
- Mission_control.js by adding some function to read Safehome from FC

```
    /////////////////////////////////////////////
    // Manage Safehome
    /////////////////////////////////////////////
    function getSafeHomePointFromFC() {
        console.log("Test");
    }
    
    function loadSafehome() {
        MSP.send_message(MSPCodes.MSP2_INAV_SAFEHOME, false, false, getSafeHomePointFromFC);
    }
    
```

The function LoadSafeHome is called on an action in the HTML file. 
The Chrome debugger indicates the following message with no errors from Chrome but from Inav : 

**FC reports unsupported message error: 8248**

The **Test** console log is written after showing that MSP.send_message is launched.

Can anyone help me understand how I can understand the MSPhelper error meaning and what I am currently doing wrong?

I have tested CLI command with Safehome, and the CLI send me the list of the 8 Safehome. Thus, I understand SAFEHOME should work. 

Regards

Arno

ArnoTlse, 2021-03-01:
Hi, 

I am going deeper. I have understood that 8248 was in fact the conversion of the value of MSP2_INAV_SAFEHOME:0x2038.
It seems that the problem is upper. In MSPHelper.js, at line 69, the dataHandler.unsupported return 1 (i.e unsupported = True) for MSP2_INAV_SAFEHOME.
The first "if condition" below works, but the second not due to  dataHandler.unsupported = 1. 

```
if (dataHandler.code == MSPCodes.MSP2_INAV_SAFEHOME) {console.log("yes");}
if (!dataHandler.unsupported) switch (dataHandler.code) {
```

Then, I have looked at MSP.js where I have found the definition of UNSUPPORTED.

```
var MSP = {
    symbols: {
        BEGIN: '$'.charCodeAt(0),
        PROTO_V1: 'M'.charCodeAt(0),
        PROTO_V2: 'X'.charCodeAt(0),
        FROM_MWC: '>'.charCodeAt(0),
        TO_MWC: '<'.charCodeAt(0),
        UNSUPPORTED: '!'.charCodeAt(0),
    },
```

It seems related to the decoding of MSP chain. Could it be an issue on SAFEHOME MSP code definition or could it be something well defined in INAV but not yet in INAV_CONFIGURATOR ?

Regards

Arno

Jetrell, 2021-03-01:
@ArnoTlse You might want to visit - https://t.me/INAVFlight
Michel Pastor is one of the iNav developers who frequents there. He may be able to point you in the right direction.

DzikuVx, 2021-03-06:
@ArnoTlse do we have to keep this issue opened?

ArnoTlse, 2021-03-10:
Hi @DzikuVx ,

Yes, If we can still keep it open. I have look slightly deeper on this in parallel on the open points discussed with @stronnag on Mission planner isssue with my first code released.
I am still facing something I don't understand. In MSPhelper.js, I have added the Case for MSP2_INAV_SAFEHOME, but when I print the data, It is empty. 

```
case MSPCodes.MSP2_INAV_SAFEHOME:
                console.log("MSPCodes.MSP2_INAV_SAFEHOME : ",MSPCodes.MSP2_INAV_SAFEHOME);
                console.log("data : ", data);
```

Log print in Chrome debugger : 
```
DataView(0) {}
buffer: ArrayBuffer(0)
[[Int8Array]]: Int8Array []
[[Int16Array]]: Int16Array []
[[Int32Array]]: Int32Array []
[[Uint8Array]]: Uint8Array []
byteLength: (...)
```

I have tested on MSP_WP_GETINFO, and the log give me a DataView(4) with data inside.

So, I surely miss something in the code where I need to indicate how to decode SAFEHOME MSP message but I don't know where.

Arno

stronnag, 2021-03-11:
Apologies if this is obvious; are you actually requesting the safehome from the FC by sending `MSP2_INAV_SAFEHOME` with a single byte in the range 1 - 8 (the safehome wanted)? Then you should get the structure back for that home. Repeat for the the rest.

This is a standard MSP pattern for lists of things, i.e. as for WAYPOINTS.

ArnoTlse, 2021-03-11:
Hi @stronnag ,

In fact, I don't really know what is all the machinery behind when I request : 

`MSP.send_message(MSPCodes.MSP2_INAV_SAFEHOME, false, false, getSafeHomePointFromFC);`

When I print the data that should be extracted by MSP.send_message, it seems it is empty.
So does it mean that something has to be modified in the MSP.send_message or maybe I miss something in understanding how MSP.send_message works ?

`send_message: function (code, data, callback_sent, callback_msp, protocolVersion) {`

How do you request info from FC by sending MSP2_INAV_SAFEHOME with byte ?

Arno

stronnag, 2021-03-11:
Can't help you with the JS side (far too difficult). However, you need a 1 byte payload where the value is the safehome you want (0-7,it's zero indexed, not 1-8 ...).

ArnoTlse, 2021-03-11:
Ok, thanks.
Yes it was that I have followed in principle. I have tried to do the same thing done for WP load. But I guess that there is a problem in the FC extraction genuine function in Inav Configurator JS files for SafeHome as the data read from FC show an empty list. And CLI extraction process seems completly different from send_message function.

ArnoTlse, 2021-03-16:
Hi @stronnag, @DzikuVx 

Just a little question more. Do you have to implement the MSP2_SET_INAV_SAFEHOME logic to read SAFEHOME ?
I guess not, but I want to be sure I'm not thinking wrong.

Are Safehomes also stored in Eprom ?

Unfortunately, I still have a 0 array when requesting 0x2038 for INAV_SAFEHOME.  Is there anything else you use in MWP to extract Safehome ? Do you know if the MSP2 structure has been changed by adding SAFEHOME on INAV C code ? Maybe something could be missed on Configurator side maybe as the 0x2038 address are quite the last one ?

What I have changed in mission_control.js

```
    /////////////////////////////////////////////
    // Manage Safehome
    /////////////////////////////////////////////
    function getNextSafehome() {
        SAFEHOME.bufferPoint.number = safehomeForSend;
        safehomeForSend++;
        if (SAFEHOME.bufferPoint.number <= 8) {
            MSP.send_message(MSPCodes.MSP2_INAV_SAFEHOME, mspHelper.crunch(MSPCodes.MSP2_INAV_SAFEHOME), false, getNextSafehome);
        }
        else {return;}
    }
    
    function loadSafehome() {
            safehomeForSend = 0;
            MSP.send_message(MSPCodes.MSP2_INAV_SAFEHOME, [0], false, getNextSafehome);
    }
```
and in MSPHelper.js :

```
self.processData = function (dataHandler) {....
case MSPCodes.MSP2_INAV_SAFEHOME:
                console.log("MSPCodes.MSP2_INAV_SAFEHOME : ",MSPCodes.MSP2_INAV_SAFEHOME);
                console.log("data : ", data);
                //SAFEHOME.bufferPoint.number = data.getUint8(0);
                //SAFEHOME.bufferPoint.enable = data.getUint8(1);
                //SAFEHOME.bufferPoint.lon = data.getInt32(2, true);
                //SAFEHOME.bufferPoint.lat = data.getInt32(6, true);
                break;
```

```
self.crunch = function (code) {
case MSPCodes.MSP2_INAV_SAFEHOME:
                console.log("SAFEHOME.bufferPoint.number : ",SAFEHOME.bufferPoint.number);
                buffer.push(SAFEHOME.bufferPoint.number+1);
                console.log("buffer ", buffer);
                break;
```

Arno

ArnoTlse, 2021-03-16:
Maybe a snapshot can help illustrate what I retrieve when send_message call for MSP2_INAV_SAFEHOME : 

[image]

ArnoTlse, 2021-03-16:
I have tested another thing:  I have replaced the code 0x2038 of MSP2_INAV_SAFHOME by another one.

```
MSP2_PID:                           0x2030,
    MSP2_SET_PID:                       0x2031,

    MSP2_INAV_OPFLOW_CALIBRATION:       0x2032,
    
    MSP2_INAV_FWUPDT_PREPARE:           0x2033,
    MSP2_INAV_FWUPDT_STORE:             0x2034,
    MSP2_INAV_FWUPDT_EXEC:              0x2035,
    MSP2_INAV_FWUPDT_ROLLBACK_PREPARE:  0x2036,
    MSP2_INAV_FWUPDT_ROLLBACK_EXEC:     0x2037,
    
    MSP2_INAV_SAFEHOME:                 0x2038,
    MSP2_INAV_SET_SAFEHOME:             0x2039
```

If I replace MSP2_INAV_SAFEHOME:0x2038 by 0x2030 as instance (for MSP2_PID), I have in console.log a non zero Array.
If I replace MSP2_INAV_SAFEHOME:0x2038 by 0x2032 as instance (for OPFLOW_CALIBRATION), I have in console.log zero Array. Same for MSP2_INAV_FW**.

Then, I wonder if there is not something needed in the MSP2 decoder in Configurator to take into account the newest address ?

Arno

stronnag, 2021-03-16:
You need to be running firmware (e.g. master dev) that supports `MSP2_INAV_SAFEHOME`; no released firmware does. If you check the MSP status / direction indicator byte, you should see '!" in older firmware that does not support this message.

ArnoTlse, 2021-03-16:
Ok thanks, that can explain things. I though it was already implemented in 2.6.0 version as mentionned in the [release page of inav](https://github.com/iNavFlight/inav/releases).  And as Safehome in CLI was working, it seems to me it was already available in the firmware I use.
How can I get the master dev firmware for my FC ? I guess I have to do it manually instead of using Configurator ?

```
2021-03-16 @ 21:57:16 -- MultiWii API version received - 2.4.0

2021-03-16 @ 21:57:16 -- Flight controller info, identifier: INAV, version: 2.6.0

2021-03-16 @ 21:57:16 -- Running firmware released on: Dec 4 2020 11:55:31

2021-03-16 @ 21:57:16 -- Board: MKF4, version: 0
```

Jetrell, 2021-03-17:
> 
> 
> You need to be running firmware (e.g. master dev) that supports `MSP2_INAV_SAFEHOME`; no released firmware does. If you check the MSP status / direction indicator byte, you should see '!" in older firmware that does not support this message.

@stronnag Might it be an idea to take the logic of this new SafeHome PR into account? #6655 
I have tested it. Its more versatile than the previous SafeHome.

ArnoTlse, 2021-03-17:
Hi,

ok, I found [it](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md) on Inav github.

I am following the instruction to build a master dev for my FC.

@Jetrell, I guess this PR will not change the MSP code structure, will it ?

Arno

stronnag, 2021-03-17:
The quoted PR is not relevant to **this PR** concerning a configurator GUI for defining geospatial spatial homes.

It would be convenient if the iNavFlight/inav#6655 configuration option could be set from the same configurator tab, but that's not essential and probably a distraction to the main task of the geo UI.

You may find find current dev binaries for popular FCs [here](http://seyrsnys.myzen.co.uk/inav_dev/)

ArnoTlse, 2021-03-17:
Hi @stronnag,

Thanks for the link. I have just finished to generate the compiled file for my FC. And youhou, I have now data extracted for SAFEHOME with my code. The counterpart is that as I have only one FC, I could not fly with my quad now but I will be able to continue on the GUI part.

Thanks a lot for the support.

Arno

stronnag, 2021-03-17:
> Hi @stronnag,
>
> Thanks for the link. I have just finished to generate the compiled file for my FC. And youhou, I have now data extracted for SAFEHOME with my code. The counterpart is that as I have [...]

AFAIK, the development builds are stable, flyable. If you don't want to build multiple configurators there are also [2.7 configurator builds](http://seyrsnys.myzen.co.uk/inav-configurator-next/); however progressively merging / rebasing your PR to configurator master may save you the pain of "big bang" re-base when you're ready for "release".

closed 2021-04-04 by ArnoTlse