iNavFlight/inav-configurator #1194: Modes tab improvements for beginners

avsaase, 2021-03-10:
Two questions I see frequently from fixed wing pilots new to INAV are: "Can I do ... with Airmode?" and "How do I enable ACRO mode? I don't see it on the modes tab". I think these misunderstandings could easily be solved by making two changes:
1. Remove airmode on the modes tab when airmode is permanently enabled. 
2. Add a bar for ACRO mode that has no channel selection sliders, and that lights up blue when no other conflicting mode is selected. I think ACRO is compatible with ALT HOLD and maybe others, so the configurator would need to be aware of this.

MrD-RC, 2021-03-10:
I think these changes would be great. I'd also +1 on removing Airmode from airplane modes completely. Permanently enabled on or off should be fine for planes. The Acro bar would also be a handy visual aid.

I'd even add a 3. To implement the auto channel select feature from BetaFlight.

avsaase, 2021-03-10:
I'm not familiar with that feature (or betaflight really). What does it do?

MrD-RC, 2021-03-10:
The drop down select box where you choose the channel. In BetaFlight the top option is Auto. You would select Auto, then move the switch on the transmitter. It would then set the correct channel in the select box.

avsaase, 2021-03-10:
Okay yeah, that sounds pretty nice.

MrD-RC, 2021-09-10:
Also, just a note. Alt Hold is not compatible with Acro. Only Angle and Course Hold are.

MrD-RC, 2021-09-10:
I've started looking in to the ACRO box thing, but have come across a slight snag. The problem at the moment is that the checks are based on if the flight controller can enable a mode, not if the channels are in range. So I can get this working, but it is currently it would only work on active modes. So RTH could be in range, but it would still say ACRO is ok.

MrD-RC, 2021-09-10:
Acro indicator done. Plus a bonus of the in-range indicator.

PR #1699 (2023-01-19): Add AUTO detect to modes page
This adds the very useful AUTO detection of the switches to the range. This is pulled from BetaFlight, where it has proven very useful.

To select the switch for the range, set the channel to AUTO, then move the switch.

Fixes #1194 

[DEMO](https://youtu.be/l9OGl_3sfUg)
closed 2023-01-19 by DzikuVx