iNavFlight/inav-configurator #1207: Setting minimum satellites in the UI

MrD-RC, 2021-03-27:
Hey guys, just wondering if this is something we should look at removing?

[image]

It's something that should rarely be used. And people in the situation where they want to, can still set it in the CLI. I just see it as a temptation that could cause people who don't know what they're doing problems. Then I guess the same could be said for the whole position estimator block. However, the number of satellites for a lock, is something that is easier for people to understand; so some are tempted to play. They don't realise that below 5 3D locks the data starts getting bad.

Perhaps we could also move the entire position estimator block to the bottom of the page, just to keep it out of immediate sight.

I'm happy to do the work. I'm just looking for thoughts from you guys. @DzikuVx @digitalentity @avsaase Anyone else?

avsaase, 2021-03-27:
I think these settings should stay. They are on a page with advanced settings and there is a big fat warning not to change them if you don't know what you're doing. I'm also not aware of any support questions caused by users changing these settings when they shouldn't. 

As for the min sats setting, the minimum value is 5 anyway. If people want to increase it a bit to get better home position accuracy there is no harm done in letting them.

MrD-RC, 2021-03-28:
Sounds fair enough @avsaase. What do you think about moving the area to the bottom of the page?

avsaase, 2021-03-28:
I think that makes sense.

PR #1212 (2021-04-01): Re-arrange advanced page
Re-arranged the advanced tuning page. This was mainly to move the position estimator to the bottom of the page. Out of sight, out of mind for those who don't need to use it. I also took the opportunity to put the fixed wing and multirotor areas together.

[image]
[image]
[image]

Fixes #1207
closed 2021-04-01 by DzikuVx