iNavFlight/inav-configurator #1237: MP Safehomes are not stored in EEPROM

DzikuVx, 2021-04-20:
Like @Jetrell noticed, after adding a new safehome and sending it to FC, setting is not stored in EEPROM and removed after power cycle.

@ArnoTlse do you think we can have a "Safe to EEPROM" command sent after safehomes or at least separate button that will trigger the save?

ArnoTlse, 2021-04-20:
Hi @DzikuVx  and  @jetrell,

I think I may have found the issue. I know that I have succeed saving safehomes in my Eprom after a switching Off & On, the Safehome was reloaded sucessfully. But I think it is because I have maybe done it several times by saving also my WP mission to EPROM.
This could be the issue. Nevertheless, in MSP, there is no equivalent to MSP_WP_MISSION_SAVE and MSP_WP_MISSION_LOAD for SAFEHOME. I guess I have to use it also for SAFEHOME ?  @stronnag, what do you think ? Is it related or not at all?

Arno

stronnag, 2021-04-21:
This could be the issue. Nevertheless, in MSP, there is no equivalent to MSP_WP_MISSION_SAVE and MSP_WP_MISSION_LOAD for SAFEHOME. I guess I have to use it also for SAFEHOME ? @stronnag, what do you think ? Is it related or not at all?

The safehomes are part of the standard settings. You use the standard `MSP_EEPROM_WRITE` to save safehomes (note this also writes all the other settings as well).

DzikuVx, 2021-04-21:
yes, the MSP `MSP_EEPROM_WRITE ` will persist safehomes after `MSP2_INAV_SET_SAFEHOME` is done

ArnoTlse, 2021-04-21:
Ok, thanks to both of you.
I'll implement this + a correction on the update of the Safehome Radius when settings are saved.

Arno

ArnoTlse, 2021-04-21:
In parallel, do you know how can I retrieve the value of safehome_max_distance and nav_fw_loiter_radius as instance. It to add another dotted circle with the radius value set to the safehome_max_distance saved in the FC at least. 

I guess it should be better to read the value from FC and not from the GUI, no? 
If OK, I guess it should be save in the FC, so I should be able to extract the info from it. Is it in MSPV2_SETTING or MSP2_COMMON_SETTING_INFO?

Arno

stronnag, 2021-04-22:
You can read any CLI `set` parameter through the `COMMON_SETTINGS` API.

ArnoTlse, 2021-04-22:
Hi,

@stronnag , thanks for the information. I have found the way to do it through mspHelper.getsetting().

@DzikuVx, I have created a PR #1242 with the correction for EPROM saving + second circle with safehome_max_distance applied as requested by @jetrell + a legend to explain the Circle around Safehome Icon.

I have tested by switching off & on my FC several time without doing anything else on the Configurator. It seems to save correctly the Safehome in EPROM now. Let me know if it is not the case on your side.

Arno

ArnoTlse, 2021-04-28:
Hi ,

@DzikuVx, do you think we can close this issue as there is a PR #1242 normally solving the pb ?

Arno

Jetrell, 2021-05-21:
> 
>do you think we can close this issue as there is a PR #1242 normally solving the pb ?

@ArnoTlse Now the fix is merged. You may as well close this.

Jetrell, 2021-05-21:
do you think we can close this issue as there is a PR #1242 normally solving the pb ?

@DzikuVx Tested many times and it all looks good. You may as well close this.

closed 2021-05-21 by DzikuVx