iNavFlight/inav-configurator #1247: Modes Tab - Show only active modes button

theproff777, 2021-04-28:
It would be really nice to be able to display only configured modes in the mode tab and flip between all possible modes and only configured modes.

I fly a number of planes and quads with iNav and often have to check what I have configured, especially if I have not flown one of them in a while and it would be nice to see only the configured ones to get a quick view as most times they should all show without scrolling.

Another option would be to show the active ones at the top, but I assume that would make the coding a little more complicated.

Tx for all the awesome work the devs put into this awesome product.

darrindevine, 2021-05-26:
I just added a PR to fix this as I also like to have only used modes visible when checking my TX switches. 

Video and Pictures of it working in my dev env below. 

https://youtu.be/xLOImnnnR5Q 

[image]

[image]

theproff777, 2021-05-26:
Awesome, that is perfect and will make it so much easier to work with the modes and also figure out what I have set.

closed 2021-06-14 by DzikuVx