iNavFlight/inav-configurator #1276: Feature Request: Offline Mode from Dump File

FPVZaphod, 2021-05-25:
Currently without a FC connected, the Configurator is useless - it's not possible to check something without connecting a Flight Controller to the Computer. 
It would be great if it's possible to open a previous dump / text file instead of a connection to a flight controller. Of course many things based on live sensor input will not work, but for checking or playing with some settings it could be  useful. 
An enhanced option would be to even enable saving when changes are done. Changing OSD setups as an example.

StuweFPV, 2022-01-09:
I would like such a "virtual" FC as well. just like in Eagle Tree Vector software when you hit ESC without having to connect the actual device. I think this would encourage users to play around or just to "check-out" inav. But I'm afraid we both have to bag that idea. Pawel wants you to buy a FC just to play around (https://github.com/iNavFlight/inav/issues/5738)
If you look at the recommended FC's it really looks to me that Matek is throwing quite some money at inav hence the encouragement to buy and no virtual device will be available. I regret this but i also understand his position. this project needs some funding in order to stay alive. we all want it to stay alive so where does the money come from? normally hardware producers act as sponsors and you can't backstab them with a virtual machine. In his position i would given the same answer.

you can always look at your diff/dump with just a notepad/text editor. i always use Winmerge where you can have 3 files open and it shows you nicely the difference of each file. edit, modify and save as. then when back on the FC just paste the new code into your inav. after a while you get quite familiar with the codes and don't need the GUI anymore. as in fact there are way more settings in the code than the GUI shows you. OSD is more difficult - there the GUI is a big benefit

FPVZaphod, 2022-01-10:
Good point, but if you like to join the hobby, you anyways need to buy a flight controller. Or - if it hits you hard - a lot of them :-)
Not just OSD is easier with GUI, also the Inflight tune settings, or serial port setup. It's not a "virtual FC" replacing a hardware one. It's just a graphical dump editor. I dont see any case, where this can replace a piece of hardware. 
I am happy to hear, that Matek supports iNav. I am sure I spent a 4 figure amount already in buying Matek stuff, and beside the fact that I am happy with it, it shows me I am supporting the right vendor.

MrD-RC, 2022-01-10:
Don’t get me wrong. I really like the idea of an offline mode. There have been plenty of times when I’ve been at work, trying to help someone out. But it would be really helpful if I could just look on iNav. I usually have to end up searching for screen shots, which are usually out of date, or not showing the right areas. 

However, this couldn’t be a part of the configurator it’s self. It would really need and external program that mimics a flight controller and serial port. Basically, everything configurator does, it gets the information from the flight controller. So the simplest way would be to just fake a flight controller with a piece of software, and just have the regular configurator talk to that. You could even have it so that you can flash iNav to the virtual FC. The biggest issue would be finding someone who wants to write the virtual FC.

StuweFPV, 2022-01-10:
I was only assuming when i said that Matek is supporting inav. I don't know this for a fact. I just look at the recommended FC's and they're all Matek and everywhere else they are omnipresent. I also spent 4 figures on their products but i feel that they are slipping. The FC's are getting worse and i had some khaki experience with their support lately which was not the case 3-4 years ago. But lets not bash on a company that's beside the point. I think we all keep our spare FC just to check something out on inav but in the case of MrD, when you're somewhere else it does not work anymore. Since it's clear that inav will never produce such a "fake" controller maybe we should get Matek to program one - even sell it. I would buy it if they had one... pick your model, gps, receiver - all virtually and connect to inav - that would be soo cool! maybe in a few years we all fly with a Raspberry and download the matek image and put that on :) being able to upgrade from a 405 to a 765 virtually :)
or with your phone - convert that into a FC - even though i would prefer a solution without cable. doesn't any of you guys know someone that programs android apps and sell it? download in playstore for 10$ :)

wx4cb, 2022-02-19:
here's a bump for this... he's been multiple revisions now and would STILL be awesome to be able to just load a diff and take a look at something quickly. heck. even just having a virtual FC like betaflight is now doing (something "we" could have BEATEN beatflight too if this had been done years ago when requested :D :D)

Phil-MC, 2022-03-05:
It should be possible to create a programmable hardware flight controller with the equivalent of a  progammable instruction set. 'Real' processors instruction sets are microcode programmed rather than hardware programmed I believe.

FPVZaphod, 2022-03-05:
My idea was not to replace hardware with other hardware - just to have a GUI to edit a dump file. No magic or calculation behind, no processing, nothing. Just a .txt instead of /dev/tty.whatever , and done :-)

wx4cb, 2022-03-05:
> It should be possible to create a programmable hardware flight controller with the equivalent of a progammable instruction set. 'Real' processors instruction sets are microcode programmed rather than hardware programmed I believe.

that's not what we're asking... think more of the betaflight "virtual FC"

Phil-MC, 2022-03-05:
I see what you are saying.  I was just mulling over alternative possibilities. I agree that it would be really useful to be able to investigate dumps without having to plug in a fc. Then anyone could constructively work with dumps without having to find the hardware. 
I usually leave a spare naked fc plugged in so I can use the Configurator. But it's not ideal.

wx4cb, 2022-03-14:
@DzikuVx sometimes its good just to be able to load a configuration to check something when something happens, like... why is this happening, you can load a diff and say , oh crap i forgot to change that. or if someone says... hey what are your xxxx settings, you can load a diff you have when the plane isnt near it and take a screenshot for them for example.

DzikuVx, 2022-03-14:
@wx4cb I get that, but to make this work the Virtual FC would have to implement at least the settings and MSP layers. With all the > 250 MSP frames FC offers. And what when new MSP frame is added? Removed? Who should remove it from Virtual FC? 
This is not something for Configurator to do. Not worth the effort.
But if someone want to make a separate project out of it, I'll support with media coverage and expertise.

closed 2022-11-26 by MrD-RC