iNavFlight/inav-configurator #1277: Online WMS/TMS without Mapproxy -> A imagery catalogue?

Matthias84, 2021-05-25:
Hi, first let me say that I'm new to UAVs but appreciate the iNAC ecosystem and your work :smile: My background is GIS and embedded systems.

The Mapproxy setup gives a great opportunity to make use of aerial imagery and maps also on your Laptop in the field.
But I guess for newcommers to this topic, the external setup might a bit complicated and parallel default OSM style a bit to generic to avoid obstacles on first flight mission plannings?

So my feature request is to allow users to add different online WMS/TMS providers. The OpenLayers widget provides already the functionality, so maybe it's enough to allow users to add a custom URL? On the other hand it might be useful to offer a catalogue of aerial imagery providers? 

The OpenStreetMap editors have a community moderated [list of map providers](https://josm.openstreetmap.de/wiki/Maps ) which offer aerial imagery as background layer. This list is parsed by the [editor-layer-index](https://github.com/osmlab/editor-layer-index), so editors can present general or location based imagery sources. Maybe this is a useful source for a location specific query?

stronnag, 2021-05-25:
You don't need maproxy; just a WMTS URI works fine.
[image]

Where locahost/mapcache points to a local WMTS

Matthias84, 2021-06-07:
Dunno why, but it didn't worked for me.

But now it works, if I switch to Map Provider to mapproxy and add e.g. _https://www.geodaten-mv.de/dienste/adv_dop?SERVICE=WMS_ with Layer _WMS_MV_DOP_ . I guess it's worth to mention, that this isn't the getabilitites URL.

Do you see any way how the management of custom different map providers might become a tip easier? For example the mentioned catalogue or a list for multiple providers which the user has to add manually?