iNavFlight/inav-configurator #1279: NAV CRUISE box always highlighted

avsaase, 2021-05-26:
The NAV CRUISE box on the modes tab is always highlighted, even when the selected channel is not in the configured range. 

[image]

MrD-RC, 2021-05-28:
It is the same for Failsafe. It does not appear to be because of the re-ordering. I removed the re-ordering code and the false highlighting still existed. 

I also tested by removing every mode except for cruise and failsafe. Both would still light up, no matter what channel I put them on, or the range selected.

Mateyhv, 2021-05-30:
Still there in RC2.

OptimusTi, 2021-05-30:
Maybe 7045 fixes it?

closed 2021-06-17 by DzikuVx