iNavFlight/inav-configurator #1290: Mission Control Default Settings display undefined values

sakisdog, 2021-06-04:
Hello, After iNav Configurator RC2 i found 2 issues:
1) The default settings values display "undefined" in the input boxes
2) I cannot click on the map to create a mission when i have no controller connected.

I don't know if this is expected behavior or not but i though i should mention it.

stronnag, 2021-06-04:
Your installation is somehow broken.
Works perfectly here:
* No dependency of an FC for mission control
* Dialogue boxes behave as expected

avsaase, 2021-06-04:
I'm having a similar problem. Once I connect and disconnect a fc I can create and edit WP's as reported in. My Default altitude and speed work as expected but the radius is undefined.

[image]

stronnag, 2021-06-04:
No such problem on 
* Linux 64
* Linux 32
* Win10 (VM)
* MacOS (VM)

Using 59d8bec / 3.0.0-RC2.

stronnag, 2021-06-04:
And the radius is undefined because you haven't defined it. Set a value, save it. Guess what you get back next time?

stronnag, 2021-06-04:
You guys are right!
Master is working
Release 3.0.0 is not.

avsaase, 2021-06-04:
Shall we close this and keep the discussion in https://github.com/iNavFlight/inav-configurator/issues/1181?

stronnag, 2021-06-04:
I think it's best to keep it here. This is essential for 3.0; some of #1181 is a wish list for future (post 3.0) work.

stronnag, 2021-06-04:
@ArnoTlse Can you please put the "can't create WPs prior to a serial connection" part to the top of your list?

ArnoTlse, 2021-06-04:
hi @stronnag ,

I have created a new branch from Master to debug it. But I do not see any problem using the master. Howerver, I reproduce the pb on the compile version 3.0.0. I have for the moment no clue of what could be the issue as it works well through NPM. 

When you say "rebuild master", it means when you create a compile version of the master ? 
How can I do that to maybe instrument a bit the code to see where could be the issue between a master running through NPM and a compile master ?

Arno

stronnag, 2021-06-04:
I have no problem with `npm start`. I get a warning for about 0.1ms that *I think says*  "profile is based in a newer nwjs" 

Maybe this is reason (and why my old master build worked and my new one fails); the build was recently updated to nw.js 0.50.3?

I'm not JS qualified in any way, this is just speculation.

ArnoTlse, 2021-06-04:
Ok, yes I have noticed this pop-up that have appeared recently, but as it really a furtive pop-up, I never reach to read what it says.

I need to have a dive in NPM and NWJS to see if I can have a NPM consistent to the version nwjs used now.

stronnag, 2021-06-04:
But I don't think NPM/NWJS is a factor. I cleaned out my repo,  reinstalled `npm` and I'm in the situation:

* Release_3.0.0 : Cannot create new WP or load mission from file prior to serial open
* Master : No problem creating WP / opening mission files

The above applies for both compiled version and `npm start`  and reinstalling `npm` has removed the momentary popup.

Beginning to doubt my sanity ....  and not being much help here.

ArnoTlse, 2021-06-04:
OK, I wonder if we reinstall the previous version of nw.js whether it is working fine or it is the same pb as now?

That you're saying is that npm start with the 3.0.0 released is also not working?

stronnag, 2021-06-04:
> OK, I wonder if we reinstall the previous version of nw.js whether it is working fine or it is the same pb as now?
> 
> That you're saying is that npm start with the 3.0.0 released is also not working?

Well I do wonder if I've just lost the plot and am not being helpful, but now I've cleaned everything out and reinstalled npm:
 * Both compiled app and npm  works with master
 * Neither compiled nor npm works with release_3.0.0

ArnoTlse, 2021-06-04:
Ok, thanks.
On my side, I have clone the `release_3.0.0` , launch `npm install` on it. But I get many error message then.

sakisdog, 2021-06-04:
To add my 2 cents here, i discovered this issue after moving from iNav Configurator RC1 to RC2. RC1 seems to work as expected. So the problem was created after code modifacations to RC2.
This is a screenshot from iNav Configurator 3.0.0 RC1
[image]

sakisdog, 2021-06-04:
Ok...So after a little digging i think the undefined values problem is because of this code:
[image]
I think that it should catch an error in case the safehome_max_distance could not be read and return the default values.

ArnoTlse, 2021-06-08:
Hi,

Really strange. This is already in the master and it works well. I do not know why it could work on master and not in the release_3.0.0.
Anyway, I will try to change it through a try... Catch to see if it helps solving the pb.

Arno

ArnoTlse, 2021-06-08:
Hi @sakisdog , hi @stronnag 

On the part of code you show @sakisdog, there is already a correction done in #1267 (older = red, newer =green), that avoid loading MspHelper when no FC is connected. Maybe that's why the Release_3.0.0 does not work if the PR #1267 has been merged in the master later but not integrated yet in the release_3.0.0 ?

[image]

It could explain why it works well with the master and not the relase_3.0.0

Arno

ArnoTlse, 2021-06-19:
Hi @sakisdog , hi @avsaase , hi @stronnag 

On the last [3.0.0 release](https://github.com/iNavFlight/inav-configurator/releases/tag/3.0.0) (7days ago), I think my previous PR has been included now. It works on my side. Points can be added on Mission control on first launch without any FC connected or with FC connected.

Are you still facing some issue on this or can we close this issue ?

Arno

stronnag, 2021-06-19:
It can be closed. Working OK.

ArnoTlse, 2021-06-19:
Ok, I cannot close it on my side. @sakisdog, if it's also ok for you, I'll let you close this Issue.

Arno

sakisdog, 2021-06-19:
It's working fine for me too. I'm closing this issue. Thank you for your help.

closed 2021-06-19 by sakisdog