iNavFlight/inav-configurator #1296: Blackbox logs saving with no file extension, .txt  missing

breadoven, 2021-06-09:
Blackbox logs no longer save with a .txt extension, they have no extension at all (only file type available is *.*). Not sure when this changed.

Is this intentional ? I guess not because Blackbox viewer won't open them without the .txt extension, thinks they're corrupt video files instead.

avsaase, 2021-06-09:
Is this a configurator or firmware issue?

breadoven, 2021-06-09:
I'm guessing Configurator since it's creating the file save dialog box that defines the save options.

stronnag, 2021-06-09:
Configurator. BBL in flash have no filename.

breadoven, 2021-08-10:
Had a play with this and found that changing `accepts `to `const accepts = '.txt';` in the following code resurrects the correct .txt file extension.

https://github.com/iNavFlight/inav-configurator/blob/b7bc81bc8daf1b4445f502a6b5c85f8e85be6bb2/tabs/onboard_logging.js#L357

However, I'm not really au fait with Java so perhaps someone who is can explain what the problem is here. I use Windows so maybe to do with that ?

closed 2021-08-12 by DzikuVx