iNavFlight/inav-configurator #1308: 3.0 : comport can't open on windows PC.

vcselvcsel, 2021-06-24:
Hello.

I flashed 3.0.0 firmware matek F411WSE using 3.0.0 configurator(win64), firmware flashing is OK, but 3.0.0 configurator didn't recognize port.

Every driver is OK, I checked my device manager, VCP driver is reconized(STMicroelectronics Virtual COM Port(COMn)) on my computer.

(I tried configurator 2.6.1 with 3.0.0 firmware target, the 2.6.1 configurator recognize 3.0.0 firmware target on CLI).

Many users in my local come up this problem.

Please check the configurator.

Thanks.

OS : Windows10 x64
target : matek F411WSE, matek F722miniSE.
configurator : win64 3.0.0, downloaded from github directly(not a chrome app).
firmware : 3.0.0 using configurator online download & downloaded from matek hompage(3.0.0)

poporbit, 2021-07-02:
issue same issue as  others.  Configurator not finding the current port.  "Manual Selection" active and  not functioning.  

PC running Win 8,  Controller flashed with INAV 3.0.0.  Configurator is 3.0.0

poporbit, 2021-07-02:
The controller board is Matek f722se

avsaase, 2021-07-03:
I often need to hit connect immediately after connecting the flight controller, like while it is still booting. If I want too long it fails to connect. No idea why that is but you can try if it works for you.

closed 2021-08-13 by DzikuVx