iNavFlight/inav-configurator #1310: BUG: Missing Glideslop feature from OSD Tab

wx4cb, 2021-06-28:
with regards to this item in the current inav 3.0 release.

https://github.com/iNavFlight/inav/pull/6939

The option to display the OSD element for the glideslope is missing.

wx4cb, 2021-06-28:
i can ebale it via CLI with 

osd_layout 2 124 1 6 V 

for example

OptimusTi, 2021-06-28:
[image]

wx4cb, 2021-06-28:
i don't see that at all....

2021-06-28 @ 16:09:37 -- Running - OS: Windows, Chrome: 87.0.4280.141, Configurator: 3.0.0

[image]

wx4cb, 2021-06-28:
[image]

wx4cb, 2021-06-28:
[image]

Mateyhv, 2021-06-29:
@wx4cb I have it the same way as @OptimusTi does.

wx4cb, 2021-06-29:
> @wx4cb I have it the same way as @OptimusTi does.

ok so what's different about your setup than mine? i even re downloaded the configurator last night  and it's still not there.

what are the requirements to get it to show lol.

OptimusTi, 2021-06-30:
> > @wx4cb I have it the same way as @OptimusTi does.
> 
> ok so what's different about your setup than mine? i even re downloaded the configurator last night and it's still not there.
> 
> what are the requirements to get it to show lol.

Not sure. The only check I see in the code is version number. Fresh install here.

MrD-RC, 2021-07-03:
Did you download an RC Configurator by mistake. This was only added 3 days before the final was released. Also, check that you haven't got multiple copies, and are running the wrong instance. 

Make sure you are using this one https://github.com/iNavFlight/inav-configurator/releases/tag/3.0.0

wx4cb, 2021-07-03:
@MrD-RC  i had re downloaded it again just to be sure. but i just downloeded another one again and ran it from a diff directory, and it is there, so something must have been weird with the install dir

closed 2021-07-03 by wx4cb