iNavFlight/inav-configurator #1311: Mission Control: Move to centre view confusing - plus some potential enhancements

MrD-RC, 2021-07-03:
## Main problem: move to centre view is confusing
This is my first time messing about with Mission Planner, and with the safehome stuff. I added and saved a safe home. But after moving the map, I wanted to view it again. So, I see a handy little button which says move to centre view. I thought that it would move the view of the map, centred on the safehome. Nope, it moves or adds a safehome to the centre of the current view.

Don't get me wrong. I can see how that would be useful. But the wording is a bit vague, and different actions can be assumed from it. Maybe change the alt text of the target icon to "Put safehome here". That is pretty obvious what it does.

## Enhancments
- **Add centre on safehome button**: This is what I thought the target icon did. It doesn't. But that would be a great feature to have.
- **Can't copy or paste with keyboard**: I tried to copy and paste coordinates in to the Lon and Lat boxes. Using _Ctrl + C_ and _Ctrl + V_ doesn't work. I had to do it the slow way, by right clicking the mouse.
- **Show the "landing" radius**: Currently there is a _Max distance_, which is the radius within which the safehome will activate. This is great. There is also a _Safe Radius_; which I can only guess is something to do with laws and distances from people or objects. However, what would be really useful is the **Loiter Radius**, as this is the path that a fixed wing will take when "landing". Having that shown will aid smarthome location placement, especially when obstacles are present in the area.

MrD-RC, 2021-07-03:
@ArnoTlse you may be the best guy to tag in this :)

ArnoTlse, 2021-07-08:
Hi @MrD-RC,

Thanks for the feedback, I will add these 3 points in the todo list. I try to push it in the preparation of the next PR.

Arno

MrD-RC, 2021-07-08:
Thanks @ArnoTlse. Great work on this 👍

ArnoTlse, 2022-03-01:
Hi,

Here are some news about two of the items mentionned here.

I have updated mission planner with a new button in the SafeHome Tab to center the map view on the selected Safehome point. I have kept the original button and rename the tooltip to indicate that it is to position the SafeHome point selected to the center of the current map view.

I have also modified the mission planner to add the Loiter Radius for Airplane (FW). It will add a yellow circle around the Safehome by retrieving the Loiter Radius from the FC (through MSPHelper). 

[image]

For multirotors and others, the default is to hide this loiter radius.
[image]

I have precised also that the initial Safe Radius is a custom one by changing the legend label.

Arno

MrD-RC, 2022-03-01:
That looks awesome @ArnoTlse thank you. The target and eye icons for the safehomes are great. No confusing what they do, even without seeing the tip.

closed 2025-01-19 by b14ckyy