iNavFlight/inav-configurator #1319: 3.0.x : Comport didn't open, depends on OS language version.

vcselvcsel, 2021-07-21:
Hello.

I've been flown with inav 1.9.x~2.6.1,

But 3.0.x configurator didn't reconize comport automatically and can't open comport.

I live in south korea, so I use korean language version windows10(x64).

One of my friend run 3.0.x configurator on english language version windows10(x64), it works.

Please fix the configurator.

Thanks.

* test condition : 
PC OS : Windows10 x64
target : matek F411WSE, matek F722miniSE.
configurator : win64 3.0.0, downloaded from github directly(not a chrome app).
firmware : 3.0.0 using configurator online download & downloaded from matek hompage(3.0.0)

jsungho, 2021-07-25:
I also use Korean Windows 64 version.
I have the same problem.

hogenejang, 2021-08-01:
same here. Can not open comport with this the lastest version. (win10 64bit (korean))
older versions are functional.

stronnag, 2021-08-04:
OS specific (i.e. Windows) issue, works fine in Linux with `LANG=ko_KR.UTF-8`.  Probably not fixable by the configurator maintainers. 

[image]

SunjunKim, 2021-12-22:
I resolved the issue by downgrading nw.js version from the current `0.50.3` to the version `0.42.2` used in 2.x.

I tested various nw.js versions, as follow:

nw.js ver / Chrome vers / Working
0.49.2 / 86.0.4240.111 / FALSE
0.47.3 / 84.0.4147.125 / FALSE
0.46.4 / 83.0.4103.116 / FALSE
0.45.6 / 81.0.4044.138 / TRUE

The latest nw.js version worked is 0.45.6, on Windows 10 Pro (20H2), language set to Korean.
This trick worked for all INAV 3.x and 4.0 versions.

If INAV configurator doesn't have an explicit reason for bumping up the nw.js version, please let them stay under 0.45.6, which will prevent the further occasion of this issue.

closed 2021-12-23 by vcselvcsel

bike797, 2021-12-23:
Betaflight configurator have this kind of issue too.
 nwbuilder version.
 - 44.6 serial port connection ok
 - 47.0 serial port connection fail
 - https://github.com/betaflight/betaflight-configurator/issues/2691

MorrisR2, 2022-01-17:
There were 65 changes between those NW versions 45.6 and 46.4.
https://github.com/nwjs/nw.js/compare/fb573512ef734f3ae085f3b56dfc8006bbd0df04...9ca47e4fc7eeefad7e2837dcaa8ca6a40ee86f9a

@SunjunKim  or someone can you try version 46.1:
https://github.com/nwjs/nw.js/archive/d9ebf24956e677658b8cbd6d0cdd7f8283384694.zip

SunjunKim, 2022-01-18:
> @SunjunKim or someone can you please try version 46.1: 
> If we figure out exactly which version change broke it, we can look at the eight or ten changes in that version.

I tested few other versions to pin out the exact version which break the compatibility.

* 0.45.6 -> chrome ver. 81.0.4044.138, working, as previously reported
* 0.46.0 -> chrome ver. 83.0.4103.116, not working
* 0.46.1 -> chrome ver. 83.0.4103.116, not working

The major change I'm suspecting is the chromium version changed from 81 to 83. Not the code from nw.js but from somewhere within chromium project.

OptimusTi, 2022-01-18:
I'm not sure if this will help, but here is a quick fix to get you going:

1. Disconnect the FC from USB
2. Delete the COM port from Device Manager
3. Reconnect the FC to USB
4. New COM port gets automatically assigned and done.

MorrisR2, 2022-01-19:
> * 0.45.6 -> chrome ver. 81.0.4044.138, working, as previously reported
> * 0.46.0 -> chrome ver. 83.0.4103.116, not working
> 
> The major change I'm suspecting is the chromium version changed from 81 to 83. Not the code from nw.js but from somewhere within chromium project.

Agreed. Looking over the changes between 0.45.6 and 0.46.0, I see nothing related to serial ports other than perhaps the Chromium version change.
https://github.com/nwjs/nw.js/compare/nw-v0.45.6...nw-v0.46.0