iNavFlight/inav-configurator #1323: Configurator build info incorrect in readme.md ?

breadoven, 2021-08-04:
https://github.com/iNavFlight/inav-configurator#building-and-running-inav-configurator-locally-for-development-or-linux-users

states using` ./node_modules/gulp/bin/gulp.js task-name` to run other tasks defined in gulpfile.js. However, this doesn't work. What does work is:

`node node_modules/gulp/bin/gulp.js task-name`

Is the current readme out of date ?

laurin-1, 2021-08-04:
Well, that is very likely, it has been last updated on Jan 29, 2020:
https://github.com/iNavFlight/inav-configurator/commits/master/README.md
For me personally, both ./node_modules/gulp/bin/gulp.js task-name and node node_modules/gulp/bin/gulp.js task-name "work", as well as npm run gulp task-name and gulp task-name (with an already installed gulp).
However, I'm not able to actually get them to properly build with the steps described in the README, the release or release-linux64 task result in the following when starting the built ./inav-configurator:
[image]
This is the same for different node versions and also regardless of how gulp was called or which gulp was used for task-name.

Using npm start works though, but that's only temporary.
The README really needs to be updated in regards to build instructions.

MrD-RC, 2021-09-10:
Building in Windows PowerShell works fine with `node ./node_modules/gulp/bin/gulp.js release`

PR #1341 (2021-09-11): Update README.md
Error in the node parts of the doc.

Fixes #1323
commit 5cc35f37 (2021-09-11): Update README.md

Fixes #1323
closed 2021-09-11 by DzikuVx