iNavFlight/inav-configurator #1324: Unable to build inav-configurator from source

laurin-1, 2021-08-09:
I'm unfortunately not able to build the configurator from source using the information provided by https://github.com/iNavFlight/inav-configurator/blob/master/README.md
The information seems to be outdated, the task release-only-linux isn't even defined in the gulpfile.js for example.
According to the documentation, npm install followed by npm run gulp release should be enough to build it, however when starting the built ./inav-configurator the result is the following:
[image]

I've tried different node versions(12.21.0,15.5.1(because it's mentioned in the window),16.6.1), the result is exactly the same with all of them. The ./inav-configurator is also a lot smaller than the precompiled one from Github, only approximately 220kB vs about 12MB. Both the release and the release-linux64 task lead to this, and it also doesn't matter how gulp is called or whether the included one or an already installed one is used.
I'm able to start the application using npm start, so I think there's either an issue with the gulpfile.js or the documentation omits an important step.

Two other things to mention:
Trying to use the release-linux64 task without prior usage of the task release results in an error(Error: ENOENT: no such file or directory, open 'apps/INAV-Configurator_linux64_3.0.1.tar.gz'), this can easily be replicated by deleting the apps folder and then trying to use the release-linux64 task.
The other thing is that the precompiled version of the configurator does not yet work with freshly compiled inav(3.1.0), the tag 3.0.1 has to be used in order to use self-built firmware with the released configurator.
[image]

I'd appreciate it if someone could update the documentation and/or fix the gulpfile.js.

alccrl, 2021-08-12:
HI,
I got same issue - Linux Ubuntu 20.04 LTS, last evening !
Build instructions are very "light".
Moreover ... I reached the same "nw.js" window only after copying dir *module-package* subdir (exact dirname may differ) from mwp-tools master dir (another tool for inav), and trying to rebuild  everything from npm install onwards.

There are many complaints on Internet about this tool chain (nw/nodejs/npm).

Alfredo

alccrl, 2021-08-12:
Errata corrige:
same tool chain of configurator is used by blackbox explorer/viewer, not mwp-tools ...
 Sorry for the confusion ...

Alfredo

alccrl, 2021-08-12:
Just an idea (not so expert in github) :
Has this behaviour to do with conflicting , one with the other, pending pull requests (e.g. #1166) ?
I included all pull requests in the source code set ..... before build.

May be someone more tunen on this repository may answer ...

Alfredo

alccrl, 2021-08-12:
After :
rm package-lock.json
npm update
npm install
npm start
inav-configurator comes out ....
but
"gulp release" and other target flavours do not produce inav--configurator as expected, only "nw.js" app appears ...

Alfredo

alccrl, 2021-08-23:
So,
I carried on today a few trials on linux-64 and I had a look at the console messages while building.
I used yarn (just to compare behaviour) together with npm, and I noticed that my "node" laptop installation was part "overwritten" by the just downloaded stuff from internet while following the readme.md instructions.

I succeded in building inav-configurator (elf) executable after resolving the windows icon missing problem.
Since I used the linux-64 task, I do not understand why the windows icon is necessary.

searchin the Internet I catched this URL:
https://askubuntu.com/questions/841847/mono-package-for-wine-is-not-installed :

What I found :
"Do the following to install wine-mono:

Download wine-mono.msi from the official WineHQ site.
Type wine64 uninstaller.
Press install from the uninstaller GUI and select the downloaded .msi package.
Done!"
it worked for me - (download same version as Your wine installation ...) and after this wine fix I was able to run gulp task again completely. : 

> gulp apps --linux64
[20:20:44] Using gulpfile ~/inav-configurator-master/gulpfile.js
[20:20:44] Starting 'apps'...
[20:20:44] Starting 'dist'...
[20:20:44] Starting 'clean'...
[20:20:45] Finished 'clean' after 398 ms
[20:20:45] Starting 'dist-build'...
[20:20:45] Starting 'build'...
[20:20:45] Starting 'build-all-css'...
[20:20:45] Starting 'build-all-js'...
[20:20:45] Starting 'build-css'...
[20:20:45] Starting 'build-receiver-css'...
[20:20:45] Starting 'build-js'...
[20:20:45] Starting 'build-receiver-js'...
[20:20:45] Starting 'build-debug-trace-js'...
[20:20:45] Starting 'build-hex-parser-js'...
[20:20:45] Finished 'build-debug-trace-js' after 161 ms
[20:20:45] Finished 'build-hex-parser-js' after 162 ms
[20:20:45] Finished 'build-receiver-css' after 175 ms
[20:20:45] Finished 'build-receiver-js' after 201 ms
[20:20:45] Finished 'build-css' after 225 ms
[20:20:45] Finished 'build-all-css' after 226 ms
[20:20:45] Finished 'build-js' after 361 ms
[20:20:45] Finished 'build-all-js' after 362 ms
[20:20:45] Finished 'build' after 362 ms
[20:20:45] Starting '<anonymous>'...
[20:20:46] Finished '<anonymous>' after 777 ms
[20:20:46] Finished 'dist-build' after 1.14 s
[20:20:46] Finished 'dist' after 1.54 s
[20:20:46] Starting '<anonymous>'...
Using v0.50.3 (normal)
Create cache folder in /home/alfredo/inav-configurator-master/cache/0.50.3-normal
Using cache for: win32
Create cache folder in /home/alfredo/inav-configurator-master/cache/0.50.3-normal
Using cache for: win64
Create cache folder in /home/alfredo/inav-configurator-master/cache/0.50.3-normal
Using cache for: osx64
Create cache folder in /home/alfredo/inav-configurator-master/cache/0.50.3-normal
Using cache for: linux32
Create cache folder in /home/alfredo/inav-configurator-master/cache/0.50.3-normal
Using cache for: linux64
Create release folder in /home/alfredo/inav-configurator-master/apps/inav-configurator/win64
Create release folder in /home/alfredo/inav-configurator-master/apps/inav-configurator/win32
Create release folder in /home/alfredo/inav-configurator-master/apps/inav-configurator/linux32
Create release folder in /home/alfredo/inav-configurator-master/apps/inav-configurator/linux64
Create release folder in /home/alfredo/inav-configurator-master/apps/inav-configurator/osx64
Update win64 executable icon
Update win32 executable icon
Zipping _locales/en/messages.json
Zipping build/debug-trace.js
Zipping build/hex_parser.js
[... 316 lines cut ...]
Zipping tabs/outputs.html
Zipping tabs/pid_tuning.html
Zipping tabs/ports.html
Zipping tabs/programming.html
Zipping tabs/receiver_msp.html
Zipping tabs/receiver.html
Zipping tabs/review.html
Zipping tabs/sensors.html
Zipping tabs/setup.html
Zipping tabs/transponder.html
[20:21:02] Finished '<anonymous>' after 16 s
[20:21:02] Finished 'apps' after 18 s

The inav-configurator executable (elf) obtained is about 14 Mbyte and works, at least on my development laptop !!!.

My impression is that a completely automated configurator build chain implies continous checks against platform tools updates and changes, and this is a very hard task to accomplish, but to add more descritpions and cases to the readme.md should be a more affordable and simple task.
As soon as I will install a fresh linux box somewhere, I will try to run the configurator on it, because my laptop is too crowded with development tools, so it does not represent a typical installation scenario.
I also will try to build under Windows. I did not yet succed... (I yet had not enough time to investigate why).

Conclusion (for now):
Somewhere in the building/execution process there is a point where errors are not reported and nw.js appears, instead of the complete configurator.

Hope this helps ...

Alfredo

laurin-1, 2021-08-23:
Amazing! Great find! Very helpful indeed, Alfredo!
I was able to successfully run npm run gulp release after installing wine-mono from https://dl.winehq.org/wine/wine-mono/ .
[image]

The readme only mentions that wine needs to be installed, not that wine-mono is required.
The release-linux64 task seems to be partially broken, it can't be run when the release task wasn't already run prior, the resulting error message is:
`> inav-configurator@3.0.1 gulp
> gulp "release-linux64"

[01:26:11] Using gulpfile ~/Git/inav-configurator/gulpfile.js
[01:26:11] Starting 'release-linux64'...
[01:26:12] Finished 'release-linux64' after 12 ms
node:events:371
      throw er; // Unhandled 'error' event
      ^

<ref *1> Error: ENOENT: no such file or directory, open 'apps/INAV-Configurator_linux64_3.0.1.tar.gz'
Emitted 'error' event on Domain instance at:
    at WriteStream.emit (node:domain:525:12)
    at WriteStream.onerror (/home/laurin/Git/inav-configurator/node_modules/readable-stream/lib/_stream_readable.js:640:52)
    at WriteStream.emit (node:events:394:28)
    at WriteStream.emit (node:domain:475:12)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: 'apps/INAV-Configurator_linux64_3.0.1.tar.gz',
  domainEmitter: WriteStream {
    path: 'apps/INAV-Configurator_linux64_3.0.1.tar.gz',
    flags: 'w',
    mode: 438,
    fd: null,
    start: undefined,
    pos: undefined,
    bytesWritten: 0,
    closed: false,
    _writableState: WritableState {
      objectMode: false,
      highWaterMark: 16384,
      finalCalled: false,
      needDrain: false,
      ending: false,
      ended: false,
      finished: false,
      destroyed: true,
      decodeStrings: true,
      defaultEncoding: 'utf8',
      length: 0,
      writing: false,
      corked: 0,
      sync: true,
      bufferProcessing: false,
      onwrite: [Function: bound onwrite],
      writecb: null,
      writelen: 0,
      afterWriteTickInfo: null,
      buffered: [],
      bufferedIndex: 0,
      allBuffers: true,
      allNoop: true,
      pendingcb: 0,
      constructed: true,
      prefinished: false,
      errorEmitted: true,
      emitClose: true,
      autoDestroy: true,
      errored: [Circular *1],
      closed: true,
[... 151 lines cut ...]
        S_IROTH: 4,
        S_IWOTH: 2,
        S_IXOTH: 1,
        F_OK: 0,
        R_OK: 4,
        W_OK: 2,
        X_OK: 1,
        UV_FS_COPYFILE_EXCL: 1,
        COPYFILE_EXCL: 1,
        UV_FS_COPYFILE_FICLONE: 2,
        COPYFILE_FICLONE: 2,
        UV_FS_COPYFILE_FICLONE_FORCE: 4,
        COPYFILE_FICLONE_FORCE: 4
      },
      promises: [Getter]
    },
    [Symbol(kIsPerformingIO)]: false,
    [Symbol(kCapture)]: false
  },
  domainThrown: false
}`

And even after the release task was run, it doesn't seem to work correctly, it only seems to generate a fraction of the files the release task generates.
So, as long as the release-linux64 task is not fixed, the installation of wine-mono is required, since only the release task seems to work correctly.
Therefore, like Alfredo already concluded, the most viable short term solution is to update the readme.

alccrl, 2021-08-24:
Thanks for testing !

I also just succeded in building under Win10, but before I had to delete mac_os and linux tasks form gulpfile, there was an exception about symbolic links during (not avoidable (?)) execution of mac_os task.
Leaving only win64 task in gulpfile, "gulp apps" task does complete and generates a zip file with inav-configurator exe file (16 mbyte).

Alfredo

closed 2021-10-12 by DzikuVx