iNavFlight/inav-configurator #1330: Configurator mission elevations don't appear to match the real world

stronnag, 2021-08-15:
Nice job on the terrain plot, looks very good, apart from one small detail. --- It does not appear to model the real world.

I tested by canonical mwp terrain analysis mission and was surprised to see I was flying through the mountains.

[image]

The indicated elevations do not seem right, WP 11 / South Barrule is below 500m. From a simple test of online datasets:
```
$ ./get_elevations -bing-key $(<bkey.txt) 54.149461 -4.669315
Using mapzen topo set
bing : 54.149461, -4.669315, elev = 470.0 in 155.481356ms
topo : 54.149461, -4.669315, elev = 476.0 in 199.898327ms
```
Wikipedia states it is 483m. The configurator plots it at over 500m, it also thinks the flight will hit the terrain at multiple points.

Neither mwp nor google-earth shows there is any contact with the terrain.
[image]

[image]

Appreciate this is a suicidal, contrived mission with insufficient ground clearance (which is it's sole purpose). 

And of course, maybe it's Google and mwp who get it wrong. Gzipped sample mission file attached.

[barrule-jump-h3.mission.gz](https://github.com/iNavFlight/inav-configurator/files/6987901/barrule-jump-h3.mission.gz)

breadoven, 2021-08-17:
Configurator shows WP11 elevation at 526m which is indeed quite a bit off.  OS map shows 483m, looks like a trig point on top of South Barrule. Configurator is using Bing maps for elevation data:

`const response = await fetch('http://dev.virtualearth.net/REST/v1/Elevation/List?points='+self.getLatMap()+','+self.getLonMap()+'&heights=ellipsoid&key='+globalSettings.mapApiKey);`

so should give the same result as MWP surely, or is there some BING setting/datum that needs tweaking perhaps ?

stronnag, 2021-08-17:
I was expecting the same results as mwp, as we both using [Bing for elevations](https://docs.microsoft.com/en-us/bingmaps/rest-services/elevations/get-elevations). mwp uses the ellipsoid model (as does GPS).

stronnag, 2021-08-17:
Interestingly, Bing states the sea level offset is 56m at the S. Barrule location. Bing WGS84 (470m) + offset 56m = 526m.

As alsolute elevation in inav is WGS84 / defined by GPS, the configruator should not be using the sea level offset (IMO); if indeed it is.

breadoven, 2021-08-17:
Would appear that ellipsoid is the correct datum to use if INAV is using WGS84 GPS heights:

https://docs.microsoft.com/en-us/bingmaps/rest-services/elevations/get-elevations#about-elevations-and-coordinate-values

The other elevation data source I use, Opentopodata.org, shows 474m at South Barrule, so the geoid value. Can't remember if it allows different datums though.

I was wondering what height a GPS would show at South Barrule which is all that matters if you want to avoid flying into it. I was thinking the GPS height will be closer to 526m than 480m until I checked a local spot that I fly from that I know has a consistent INAV MSL altitude of around 275m. Adding a WP here in Configurator returns an elevation of 305m, significantly higher. So I changed the datum used by Configurator to sealevel from ellipsoid and it shows ... 252m. So neither match the actual GPS height, both out by more than 20m. Having said that the ellipsoid would appear to overread perhaps which does at least provide a safety margin.

stronnag, 2021-08-17:
> I was wondering what height a GPS would show at South Barrule which is all that matters if you want to avoid flying into it.

The only one I have. Somewhat ancient (portable WBT-201 GPS), quite a good device for its time but probably no match for a modern Ublox.

[image]

breadoven, 2021-08-17:
Does seem that the actual GPS reading when set to WGS84 is closer to reality than the theoretical heights, ellipsoid or geoid. The local flying site I mentioned above has an OS contour height of 280m so close to the actual INAV MSL GPS value of 275m.

Checked Configurator at South Barrule with sea level datum and it shows 470m which is closer than the ellipsoid value I guess. Also put a WP in the sea near the Isle of Man with the ellipsoid datum set and it shows 55m, the SL offset.

Part of the inaccuracy is perhaps down to the DEM model resolution. Seems to be 10m in the US, only 90m elsewhere which could make a bit of a difference on uneven ground.

https://docs.microsoft.com/en-us/bingmaps/rest-services/elevations/get-elevations#earth-models-and-zoom-level

Maybe Configurator needs an option to select the datum or even use the average of the datums since that would seem to be closer to reality.

stronnag, 2021-08-22:
closing, inav ublox uses the GPS' altitrude_msl, so the configurator is technically correct, albeit somewhat adrift of reality due to the  rather large correction grid area used. 

There is no good answer; caveat user. Give yourself plenty of margin, check with a survey map.

closed 2021-08-22 by stronnag