iNavFlight/inav-configurator #1338: BBL download has no extension

wx4cb, 2021-09-06:
Issue: when downloading a log file from dataflash, the configurator fails to save it with a .TXT extension. I know this is trivial, but the blackbox explorer doesn't recognise the log without it.

Fix: Add a .txt extension when saving blackbox logs from a dataflash chip

stronnag, 2021-09-06:
Already fixed by #1325

wx4cb, 2021-09-06:
@stronnag which version as i'm using 3.4.0 that i just downloaded

stronnag, 2021-09-06:
github master. If you want a pre-release, then as ever [my master/next stash](http://seyrsnys.myzen.co.uk/inav-configurator-next/), supports 3.0.x and 3.1. If you're on 3.4.0, then you're kilometres  ahead of the rest of use :-)

wx4cb, 2021-09-06:
i meant BBE 3.4.0 and im on 3.10 for inav.  :D

stronnag, 2021-09-07:
You're correct (as usual). The download folder is filtered by `*.txt`; however `.txt` / `.TXT` is not automagically appended to the generated file name; the user has to append it.  So #1325 does not appear to fix your issue.

[image]

Other OS may behave differently ....

breadoven, 2021-09-11:
Maybe the easier solution is to just make the .txt extension part of the file name by default. I think the diff file save does the same, no separate extensions necessary in that case.

stronnag, 2021-09-11:
yes, that would be the best, probably .TXT for backwards compatibility (and case sensitive OS).

closed 2021-09-13 by DzikuVx