iNavFlight/inav-configurator #1339: Inav 3.1 doesnt show crossfire receiver

Jonathantomlin8, 2021-09-08:
I cant get inav 3.1 receiver page to work. I am using crossfire and the same board matek 411 wse with the same crossfire receiver connected. It functions perfectly on inav 2.6 but doesnt work on 3.1. Please help

DzikuVx, 2021-09-08:
Any screenshot?

Jonathantomlin8, 2021-09-08:
Hi Pawel

Thank you for emailing me I am a great fan of yours on youtube!

I have done a controlled experiment to demonstrate the receiver not showing on INAV 3.1 here's what I did:
1 Using a new Matek F411-WSE board flashed the latest 2.6 version
2 Using INAV 2.6.1 configurator connected and everything was working including the receiver page with movements showing correctly (please see attachment showing screenshot)
3 Without moving or unplugging anything opened INAV configurator 3.1 and flashed 3.1 onto the board
4 connected to INAV went to receiver page nothing was working (see attachment 2 screenshot)

I have replicated this experiment with another Matek F411-WSE board and got the same results. I am very pleased that I can use INAV even if it's only 2.6 my models
are a ZOHD drift and an Eachine razor but I thought you would want to know about this bug and when it's fixed, I will upgrade to 3.1.

Good luck with the fix and keep up the good work.

Best wishes

Jonathan

OptimusTi, 2021-09-08:
I have the same setup without any issues. Screenshots didn't make it here. Post a diff?

MrD-RC, 2021-09-10:
Do you mean 3.1 or 3.0.1?

I've not had any problems with Crossfire either. Have it on an F411-WSE too.

Jonathantomlin8, 2021-09-10:
So does the fact you haven't had problems prove there is no issue when I have sent you evidence of the problem I have had?

MrD-RC, 2021-09-10:
@Jonathantomlin8 the screenshots don’t show up on here when you reply to an email. You also haven’t confirmed which version you’re actually using. 3.0.1 is the current stable release. 3.1 is the current development version. 

You have said that you are having issues. But other people not having issues tend to lean towards  a setup problem rather than a firmware problem. Things have to be ruled out.

Jonathantomlin8, 2021-09-11:
Hi Darren

I have copied you in on the original email plus attachments. The set up is not the issue as I didn't do anything to the flight controller
I left it plugged in and did the experiment with 2.6 which works then 3.0.1 which doesn't show anything on the receiver page. I had the same results on another new flight controller (same model)
so it is not the flight controller or set up.

I hope this helps

Jonathan

MrD-RC, 2021-09-11:
@Jonathantomlin8 It just reposted the same email message on GitHub. For images, the best thing to do is go to the issue on GitHub and add them to a reply there, not via email. Emails you get related to an issue will have a “View in GitHub” link at the bottom. If it’s anything other than just text in your response, it’s better to click the link and reply on the site. 

Just installing a new version over the top of the old does not necessarily mean that it’s not a setup issue. Things can change between versions. 

I’ve just gone through the update procedure on an F411-WSE to 3.0.1. That’s a full chip erase update, as recommended. CRSF appears on the receiver page and could be selected. After importing the diff, I had full control on the receiver page. All bars moved as expected.

Jonathantomlin8, 2021-09-12:
Can you explain to me what importing the diff means as I have like you done a full erase CRSF appeared but no control of the page maybe importing the diff is what I am missing.

Thanks
Jonathan

MrD-RC, 2021-09-12:
@Jonathantomlin8 It’s the standard iNav upgrade path. On the old setup you go to the CLI and type **diff /all** and hit the __enter__ key. You save the output to a text file or copy the script. With major versions (2.5 to 2.6 for example) you always need to check with the release notes to see if you need to modify the diff, ready for the new version. Then flash the new version with full chip erase. Once flashed, connect to the flight controller, select the correct default from the popup. Once that reboots, go to the CLI, paste in the diff, press __enter__, then type **save** and press __enter__. After that reboot, your flight controller should be set up the same as with the old firmware, minus any manual changes you may have to make, specified in the release notes. 

If you’ve just performed a full chip erase. Did you re-setup the ports? You will need to make sure that Serial RX is the only thing on UART 2 and that SoftSerial 2 has nothing assigned to it, if you are using the default MATEKF411SE target.

Jonathantomlin8, 2021-09-12:
Thank you for the explanation. I wonder if someone had never used iNav before and was starting with 3.0.1 with no diff if they would have the same experience as I did. Maybe as an experiment you could try doing it without the diff and see if the receiver page still works

I set up the ports from the Matek manual this involved switching something off on the left (UART 2?)and turning on the GPS I think the receiver was already on. I haven't used the soft serial 2.

Thanks for your help. I will try again using the diff.
Cheers Jonathan

OptimusTi, 2021-09-12:
Post the zipped file of the diff info from 2.6 with working setup so we can take a look at it please.

MrD-RC, 2021-09-12:
@Jonathantomlin8 If you don't have the diff, you just set up iNav as a fresh start. It still works fine like this.

On the ports page you need to have 
UART 1 set to GPS
UART 2 Serial RX on and nothing else.
SoftSerial 2 should have nothing assigned.

[image]

OptimusTi, 2021-09-12:
softserial 2 should be disabled on the configuration page, right?

MrD-RC, 2021-09-12:
No, just not having anything enabled on SS2 should be fine. On the configuration page you only have the option to disable SS completely, not individual ports. You should be able to have GPS and CRSF on the UARTs and something like SmartAudio on SS1.

MrD-RC, 2021-10-09:
@Jonathantomlin8 is this resolved now?

Jonathantomlin8, 2021-10-09:
Hi Darren, running it with INAV 3.0.2 was the solution so yes, it is now resolved. Thank you, Jonathan