iNavFlight/inav-configurator #1349: I can't use the BNO055 sensor.i want use IMU2.

ZJyche, 2021-09-19:
I have a drone with kakuteF7 flight control.
I want use a imu2 sensor.

I bought a BNO055 sensor from Taobao.com，it's GY-BNO055.
https://item.taobao.com/item.htm?spm=a230r.1.14.16.70316963qHzWqG&id=541798409353&ns=1&abbucket=9#detail

I connect the SCL pin of the sensor to the SCL pin of the flight control
and SDA to SDA,VIN to VIN,GND to GND.

but if I connect bNO055, all i2C bus devices will not work. (barometer,sonar...)

Can someone tell me how to use the BNO055 sensor?

BNO055         kakuteF7
     |                        |
  SCL  ------->  SCL 
  SDA -------> SDA
  VIN  ------->  5V
  GND -------> GND

DzikuVx, 2021-09-19:
There are 2 soldering jumpers on the most of bno055 boards. They boths have
to be closed for I2C mode

ZJyche, 2021-09-20:
Thank you for your reply. Your reply speed surprised me.

I checked with the seller of the sensor.
He said that the default communication mode of the sensor is i2C.
He says the sensor only needs to be connected to SDA,SCL,5V,GND pins.

These are the pins of my senso,And the connection to my flight control:

BNO055 all pins / My flight control
VIN -------------> 5V
GND ------------> GND
SCL/RX ---------> SCL
SDA/TX ---------> SDA
ADD (No connection)
INT (No connection)
BOOT (No connection)
REST (No connection)

And I don't currently have any short-circuit sensor pins.

I'm sorry to bother you, but I know very little about this sensor. Could you please give me more hint? Thank you very very very much!

DzikuVx, 2021-09-20:
The jumpers are those small pads near XYZ mark. Are they soldered together or not?

ZJyche, 2021-09-20:
Thank you very much for your reply. 

I noticed two small pads near  XYZ mark.they are not together.
Do I need to connect them?

ZJyche, 2021-09-20:
Sorry, the pad seller near XYZ said it was connected. Now it is in I2C mode

ZJyche, 2021-09-21:
I mean, sensors in I2c mode still don't work and cause all I2c bus devices to not work

DzikuVx, 2021-09-21:
Then it seems like there is something wrong with the hardware or wiring, as the board breaks the whole i2c bus

IvanBiv, 2022-07-17:
@DzikuVx hello
In INAV 5.0 and 4.1 BNO055 (IMU2) used for navigation?