iNavFlight/inav-configurator #1420: Blackbox tab times out with Matek F405mini

rickgode, 2021-12-25:
Is anyone else having issues trying to access the blackbox in inav configurator 4.0 on a Matek f405 mini?  I have two, and both just time out.  I also have a Matek F405SE and a FlywooF411V2 and I can access the blackbox menu in configurator just fine.  Thoughts?

Ricky

rickgode, 2025-01-22:
https://youtu.be/X-arqdeeZHU

Ever since inav 4.0 the blackbox tab hasn't worked with any of my MatekF405 mini flight controllers.  I updated to inav8 today and it is still a bug.  Although the Blackbox (MSC) button seems to work on inav8, I couldn't access my blackbox logs. (I suspect it was because my dataflash was full)

On a whim, I downgraded my MatekF405 mini back to inav 3 (the blackbox tab worked fine) and I deleted the Dataflash.  Then I updated back to inav8 and before I restored my diff I went into the blackbox tab and enabled blackbox logging.

Then I went into the CLI tab and set the blackbox device = SPIFLASH and tried the blackbox tab and still NOTHING...

But when I went to the CLI and pressed the Blackbox (MSC) button, I was able to access my DataFlash!  I went outside and armed/disarmed real quick so that I would have some logs and I was able to access them via MSC and even delete them.

I hope this helps anyone that happens to be still running the MatekF405mini (it's a great FC!)

b14ckyy, 2025-01-22:
I closed the ticket since it was 3 years old with no reply and no further reports. 
But looking at the other linked ticket it was also never solved. I will reopen this one and move it over to INAV (since a missing flash is not a configurator issue). 

This is probably a Target issue that got forgotten.