iNavFlight/inav-configurator #1421: Crosshair in OSD not in middle for dji, not movable

dvdboom, 2021-12-26:
Unusable crosshair in OSD as it it posioned to the left in DJI goggles. Many things look wrong in configurator osd compared to 16:9 dji goggles but can be moved to fix. This one is stuck at the wrong spot.

MrD-RC, 2022-01-08:
The configurator matches up perfectly for analogue OSD. There must be an issue with the DJI stuff for the alignment. I've just checked the CLI, and there is nothing to move the crosshair that might have helped.

I don't have DJI, so can't confirm the issue. And DJI's lack of ability to record the OSD in the DVR also makes issues like this frustrating. Hopefully, DJI will do the right thing and implement the canvas based OSD. That would sort out a lot of the issues with DJI.

MrD-RC, 2022-11-26:
Closing due to WTF OSD, which does the work DJI should have done for a proper OSD implementation.

closed 2022-11-26 by MrD-RC