iNavFlight/inav-configurator #1426: Parachute not working 4.0

wx4cb, 2022-01-04:
The parachute icon for some reason doesnt indicate a failsafe and the fc doesnt always go into failsafe when used with a tbs sixty9 vtx/rx ckmbo.

Crsf is used between the rx and the f765 wse, fresh build with the latest inav release.

The rx is going into failsafe and the fs mode is set tk cut.

The OSD shows !FS! In the mode area

wx4cb, 2022-01-04:
[image]
[image]

OptimusTi, 2022-01-04:
Do you have another CRSF Rx to test with?  Test two planes on INAV 4.0 - Diversity RX, Nano RX. Latest TBS firmware.
[image]

wx4cb, 2022-01-04:
that's what im working on now... i know it's fine with every other crossfire rx i have.... trying to see if it's specific to the 69 as this is the first one i've used. i'm updating a new nano rx now

wx4cb, 2022-01-04:
it's just weird how the osd shows it's in FS but the configurator doesnt

MrD-RC, 2022-01-08:
Configurator will only do what the firmware tells it to, so I believe the issue will be firmware side. But yes, it is strange that there is a mismatch somewhere.

closed 2022-02-19 by wx4cb